robotology/mesh-support #247: Left Shoulder not moving

AlexAntn, 2016-05-31:
Shoulder won't move on startup on iCubPlymouth01, robotInterface keeps printing:
`[ERROR]sharedcan [1] board 1 ABS ENCODER ERROR AXIS 1`

This problem came up after we had to replace cable 4 on the left shoulder. 
yarpMotorGui says that joint 1 has an hardware fault, and the angle of this joint does not update even when moving the arm.

I checked the cables both on the back and on the arm, and no cable was visibly broken.

Any other possible debug we can run?

julijenv, 2016-05-31:
Hi @AlexAntn ,
Could you explain me how you proceed to restore the cable 4?
what did you unmount? Did you remove the upper arm cover, by any chance?
If not, just check this cable.
[image]
Let me know,

AlexAntn, 2016-05-31:
Hello @julijenv ,

We did not remove the upper arm cover when replacing the cable, we only removed the torso front-cover.

I checked those connectors, didn't see anything obviously out of place. In either case, i did take a picture:

[image]

Thanks for the reply!

julijenv, 2016-06-14:
Hi @AlexAntn ,
could you please take a look at the sensor at the level of the shoulder 
AE2 on the photo below.
[image]
I guess this one is broken, most probably one of the wire is unsoldered?.
anyhow, please try to get in touch with icub.cupport on skype, thx

AlexAntn, 2016-06-16:
Hello @julijenv ,

I checked the sensor in AE2 and didn't see any broken cable. I opened the arm covers to see if the cable could be broken half-way, and i couldn't see any signs of it either.... 

This being said, the cable could be broken half-way, beneath the rubber cover...

I'll try to get in contact with the icub support on skype, thanks!

julijenv, 2016-06-23:
Hi @AlexAntn ,
any news?
Could you try to remove the connector I told you that was probably broken and check (when robot has the red butoon pressed) if the message is still the same.
send me the log file.
thx in advance

AlexAntn, 2016-06-23:
Hello @julijenv ,

No news yet, as i haven't been able yet to find the time to contact icub.support on skype.

In any case, i ran the yarprobotinterface twice, first with the cable still plugged in (nothing changed) and then with the cable plugged out (the top-most one on the left side, as indicated on your image).

1) With the cable still on: [OutputCableOn.txt](https://github.com/robotology/icub-support/files/330308/OutputCableOn.txt)

2) with the cable unplugged: [OutputCableOn.txt](https://github.com/robotology/icub-support/files/330309/OutputCableOn.txt)

From what i saw on the output, it's a different cable/board (the main difference was that, with the cable unplugged, both board 1 and board 2 were showing errors, as opposed to just board 1 with the cable still plugged).

Thanks for the reply!

julijenv, 2016-06-23:
Hi @AlexAntn ,
could you reup the file with the connector unplugged coz actually you put twice the same file.
c u soon

AlexAntn, 2016-06-23:
Hello @julijenv ,

I'm so sorry, here it goes:

2) with the cable unplugged: [OutputCableOff.txt](https://github.com/robotology/icub-support/files/330483/OutputCableOff.txt)

Thanks!

julijenv, 2016-06-27:
Hi @AlexAntn,

ok so now I understand my error, it is for sure AE2 which has a problem but the connection is done at the level of the board 1B0 (the one below and the connector is labelled S2 - in the inside part of torso).
do continuity test and try to understand which cable is broken.
good luck, let me know

AlexAntn, 2016-06-29:
Hello @julijenv ,

I was going to perform the continuity test today on the cable S2, but as i was reaching the cable, i noticed a thin white wire loose (see image), which is probably the cause of the error.

[image]

[image]

(sorry for the low quality pictures)

This being said, the connector is in a really hard place to remove, as it is blocked by the metal cillinder (to attach the back cover of the iCub, see picture).

How can i remove the connector? is it possible to remove the blocking cillinder? 

Thanks!

julijenv, 2016-07-01:
Hi @AlexAntn ,
Please unscrew the standoffo cover (mettalic cilinder) and rewire the connector. Let me know.

julijenv, 2016-07-04:
Hi @AlexAntn ,
any news?

AlexAntn, 2016-07-04:
Hello @julijenv ,

I still haven't been with the iCub since friday, i will let you know as soon as we get some news!

Thanks!

AlexAntn, 2016-07-22:
Hello @julijenv ,

We finally managed to crimp the wires back. Now everything is working fine! :)

Sorry for taking this long to provide an update, and thanks for the help!

closed 2016-07-22 by AlexAntn