robotology/mesh-support #248: Several problems

amaroyo, 2016-05-31:
A small list for the red iCub to be fixed (please! :)

1) Skin failure: The whole left arm seems to be disconnected; some of the "triangles" of the torso and right arm are not working; right arm index and pinkie finger tips don't work.

2) There is a loose cable in the left wrist giving problems while bending.

3) Missing a silicon ring on the left thumb.

4) The tip of the left index finger is detached. 

5) Right ring finger doesn't work properly. The distal joint does not extend fully. 

Thank you!!
Alex.

julijenv, 2016-05-31:
Hi @amaroyo ,
I took note of it, we will do it ASAP return from outside businees (on friday or next monday)

amaroyo, 2016-06-07:
So, @julijenv 

rewriting the list with the things already fixed:

1) Right ring finger doesn't work properly. The distal joint does not extend fully. [**Postponed**]
2) Left hand finger (Joint 7) is giving hardware problems. It's shaking, opening and closing very fastly, all the time. Fixing the PID doesn't work. [**New**]
3) Skin:
3.1) Left arm totally off
3.2) Right hand: thumb and pinkie fingers are not working
3.3) Right arm and forearm are giving problems. See attached picture.

Thank you!!!
Alex.
[image]

reafrancesco, 2016-06-08:
Please @julijenv mind the fact point 2 of the list is new issue.

**2) Left hand finger (Joint 7) is giving hardware problems. It's shaking, opening and closing very fastly, all the time. Fixing the PID doesn't work. [New]**

This is due to anomalous control of the joint related more to electronics than hardware, probably.

reafrancesco, 2016-06-08:
update:
In relation to the list provided by @amaroyo 

1) Right ring finger doesn't work properly. The distal joint does not extend fully. [Postponed]
~~2)~~ ~~Left~~ ~~hand~~ ~~finger~~ ~~(Joint 7)~~ ~~is~~ ~~giving~~ ~~hardware~~ ~~problems.~~ ~~It's~~ ~~shaking~~, ~~opening~~ ~~and~~ ~~closing~~ ~~very~~ ~~fastly~~, ~~all~~ ~~the~~ ~~time.~~ ~~Fixing~~ ~~the~~ ~~PID~~ ~~doesn't~~ ~~work~~.  [FIXED]
3) Skin:
3.1) Left arm ~~totally~~ **only is off** 
3.2) Right hand: thumb and pinkie fingers are not working
3.3) Right arm and forearm are giving problems. See attached picture

julijenv, 2016-08-29:
Hi @reafrancesco ,
could you update this issue please or close it if ever.
thx

julijenv, 2016-10-04:
Everything was alright the last time I checked it before you go in the erzelli, no?
could you please close this issue or give me feedbacks.

closed 2016-10-14 by julijenv