robotology/mesh-support #249: Joints 4/5/6/7/8/9/10/11/12/13/14/15 of right_arm go in fault/idle during or immediatly after startup

traversaro, 2016-06-01:
##### Description of the failure
During startup or immediatly after startup, the joints of the right arm after the 4 go to fault/idle. 

##### Detailed conditions when the failure occurred
I will comment this issue for each occurrence on the problem. : )

traversaro, 2016-06-01:
## Conditions
We startup the robot successfully, then after some minutes the hand fall. 

## Log 
https://gist.github.com/traversaro/0ab25c4bdd71aabd96baa7caf3143deb#file-log_1_jun_2016_lumix-txt

## Joints situation
[image]

traversaro, 2016-06-01:
## Conditions 
We started the robot, and the joints 8 => 15 went in fault during calibration.
After some minutes also the joints 4 => 7 went in fault. 

# Joint conditions 
After startup:
[image]

After some minutes:
[image]

## Log 
https://gist.github.com/traversaro/d54e57868a6b290d8dad8e4da93b42ee#file-log_1_jun_2016_lumix_test2-md

traversaro, 2016-06-01:
@julijenv found the problem: on cable 4N1 the two CAN cables (black and white) where exchanged:
[image] 

@julijenv fixed it and now everything seems to be working fine.

traversaro, 2016-06-01:
Actually there are still problems, and @julijenv will check again friday morning.

traversaro, 2016-06-02:
Current situation: with the `r_upper_arm` link without cover, starting the robot will always result in the Joint 7 of the right_arm (`r_hand_finger`) to be in Hardware Fault (I guess there is some board detached). 

Occasionaly there is also an hardware fault on the joint 1 of the right_leg (`r_hip_roll`), but this failure does not seems to be related.

traversaro, 2016-06-14:
Update: at the moment, the only joint that is consistently in hardware fault is the joint 7 of the `right_arm` (`r_hand_finger`).

davidetome, 2016-07-21:
the motor was shorted with the chassis. Replaced -> Fixed

closed 2016-07-21 by davidetome