robotology/mesh-support #254: canBoard 10.0.1.6 CAN 2 not responding

fjandrad, 2016-06-10:
Please fill in the following two sections:

When starting yarprobotinterface the following errors are observed :
55826		59415,021213		ERROR		 from BOARD 10.0.1.6 (EB6) @ 72s 588m 52u: CAN discovery after 2999 ms has detected 1 missing eobrd_strain boards in CAN2:
55827		59415,021287		ERROR		1 of 1: missing eobrd_strain BOARD 10.0.1.6:CAN2:13
55828		59415,021306		ERROR		embObjStrain::open() has an error in call of ethResources::serviceVerifyActivate() for BOARD EB6 IP 10.0.1.6 
 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 72s 588m 65u: (code 0x05000009, par16 0x010d par64 0x0000000001000000) -> CFG: EOtheSTRAIN cannot be configured. can discovery fails. can address in p16, prot and vers in p64 lower 32 bits + .
 from BOARD 10.0.1.6 (EB6), src CAN2, adr 0, time 76s 754m 10u: (code 0x00000019, par16 0x1b01 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .

##### Detailed conditions when the failure occurred
Just starting the robot first time in the day, and robotinterface fails to start due to the previous mentioned errors. From the outside we saw a disconnected cable in the left leg which and thought it was the reason of the errors so we took out the covers connect the cable and reattach the covers but the error remained the same and no new errors where discovered. We later tried to see if we could connect through canLoader to the can directly but no board was found although CAN1 in same address is working properly.

[image]

**Remark**: if you are not a daily icub-support user (e.g. newbie), please read carefully the guidelines for contributing linked at the top of this page. In particular, the name of your robot must be always indicated: either using labels if you're already a contributor, as per standard, or by specifying it while detailing the issue.

fjandrad, 2016-06-10:
cc @traversaro

julijenv, 2016-06-14:
Hi @traversaro ,
has been restored, can you confirm and closed the issue?

traversaro, 2016-06-14:
The sensor is working fine, thanks.

closed 2016-06-14 by traversaro