robotology/mesh-support #255: Not able to boot on iCub image 6

ahoarau, 2016-06-13:
##### Description of the failure

The iCub 6 seems to be built for an x64 CPU :  
```
This kernel requires x86_64 cpu, but only detected an i686 cpu, unable to boot. 
Please use a kernel appropriate for your CPU
```

##### Detailed conditions when the failure occurred

Thanks to your tuto on issue https://github.com/robotology/icub-support/issues/152 (I think the issue is the same btw ), @jeljaik and I managed to plug a monitor on Paris01 pc104.

We'll try later with icub image 5, but i'm pretty sure it'll be the same error.

jeljaik, 2016-06-13:
As suspected by @ahoarau we have the same problem with image 5.0 on `iCubParis01`. The exact message when booting is: 

```bash
Loading /live/vmlinuz.....
Loading /live/initrd.img...............ready
This kernel requires an x86-64 CPU, but only detected an i686 CPU. 
Unable to boot - please use a kernel appropriate for your CPU
```

jeljaik, 2016-06-14:
@mbrunettini Can you help us please?

mbrunettini, 2016-06-14:
Latest images are built only for x64 processors.

jeljaik, 2016-06-14:
iCubParis01: 002-03-04-04-04
iCubParis02: Not sure where to find it. @rlober do you?

rlober, 2016-06-14:
@jeljaik No idea. I am guessing it is on pc104

jeljaik, 2016-06-14:
On iCubParis01 it is as seen in this picture:  
iCubParis02 does not have it though ...

[image]

rlober, 2016-06-14:
@jeljaik That is because iCubParis02 is cursed.

jeljaik, 2016-06-15:
@mbrunettini Should we assume that the conclusion is to stick to the 4.0 image forever and ever?

jeljaik, 2016-06-15:
If that is the case, then there's something really wrong going on here, because in [these](http://wiki.icub.org/wiki/How_to_burn_Linux_pc104_image_(version_prev._5.0)) for < 5.0 images, the link takes us to this: 

[image]

Where clearly there's no < 5.0 image.

😟 ❓:question:

jeljaik, 2016-06-15:
Thanks for the 4.0 images! I just saw they've been added.

jeljaik, 2016-06-16:
So, we successfully managed to boot with Image 4.0, **but only with the static one**. We couldn't get the **dhcp** image   [here](http://www.icub.org/download/pc104-images/) to boot. We tried both EXT4 and FAT32 formats and checked with `md5sum` the uncompressed files. We also made sure to use the `DEVICE_NODE` rather than the partition. Still, the `dhcp` image shows the following message when connecting a monitor to `pc104`.

[image]

lornat75, 2016-06-17:
iCubParis01 mounts the old can bus interface with requires images 1.8 (which have been restored in http://www.icub.org/download/pc104-images/). It is not only a problem of architecture (32 bits): the can bus driver is available only for the Linux kernel version shipped with images 1.8. More recent versions may boot, but will not load the can bus driver.

iCubParis02 on the other hand should work with images 6.0.

jeljaik, 2016-06-17:
Thanks @lornat75 It's now clear. We'll proceed accordingly.

jeljaik, 2016-06-20:
Hello @lornat75 . We went back to the 1.8 image but with the latest `yarp` we get the following when compiling on PC104

```
/usr/local/src/robot/source/yarp/src/libYARP_OS/src/BottleImpl.cpp: In member function 'virtual void yarp::os::impl::StoreInt64::fromString(const yarp::os::ConstString&)':
/usr/local/src/robot/source/yarp/src/libYARP_OS/src/BottleImpl.cpp:722: error: 'strtoll' is not a member of 'ACE_OS'
make[2]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/BottleImpl.cpp.o] Error 1
make[1]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/all] Error 2
make: *** [all] Error 2
```

On PC104
```
icub@pc104:~$ dpkg -l | grep libace
ii  libace-dev                              5.4.7-12                             C++ network programming framework developmen
ii  libace5.4.7c2a                          5.4.7-12                             C++ network programming framework
```

jeljaik, 2016-06-20:
Ref: https://github.com/robotology/yarp/issues/807

julijenv, 2016-10-04:
Hi @jeljaik , I think you can close it no? :D

jeljaik, 2016-10-04:
Oops, yes. Had closed only the referenced issue robotology/yarp#807

closed 2016-10-04 by jeljaik