robotology/mesh-support #257: Power issues

kt10aan, 2016-06-15:
Our iCub hadn't been turned on for some time now (two months or so)

Today we turned it on, CPU first. PC104 booted normally, sshed into it and then turned on the motors.
As soon as we turned on the motors, the iCub lost power completely (both CPU and motors)

Then the switches would not turn on. We had to setup the power supplies again. (see https://github.com/robotology/icub-support/issues/187)

However, the second power supply works only if it is set to ``Local``. If we set it to ``Remote``, the ``Output`` of the second one turns off.

Now, with both power supplies in ``Local``, we first turn on the CPU, boot into ``pc104``, then when we try to turn on the motors, total loss of power happens.

Is it an issue with the settings of the power supplies? (Hopefully...)

P.S. Also, I do not know if it relevant, but the LEDs of the power switches do not turn on.

julijenv, 2016-06-15:
Hi @kt10aan ,
please send a picture of the settings for both power supplies, 
as well as the overvoltage and overcurrent settings for each of us, you may have reduce the overvoltage by inadvertence.
send the pix,
thx in advance

kt10aan, 2016-06-20:
Detaching the serial connector, powering off the power supplies and dialing in the settings again, they work but in Local mode both. LEDs on power switches are fine too.

Too many photos to post for the settings but here are the numbers for both:

Top:
Voltage: 13.25V
UV: 11.50V
OV: 15.00V
Power: 500W
Current: 20.00A

Bottom:
Voltage: 38.80V
UV: 24.00V
OV: 47.50V
Power: 1.5KW
Current: 60.00A

Last time we had similar issues with the power supplies was (again) after some period of not using the robot.

As I said, now both working fine, but only in Local mode.

julijenv, 2016-06-22:
Could you please tell which robot you r using? (label it).
which power supply are u using at least one picture could be from a great help btw.
thx in advance

mulkkyul, 2016-07-04:
Hi @kt10aan  and @julijenv ,
Any updates on this issue? 
We (iCubDaejeon01) had a similar issue (https://github.com/robotology/icub-support/issues/264) and we could solve it by changing CPU OV. (from 14.5V to 15V). If you have some updated information, could you post it? Thanks!

kt10aan, 2016-07-06:
As I mentioned in my previous post, our CPU OV is 15V already. It didn't help. I took it back to 14.50.

The bottom one will have its Output to Off, if it is in Remote mode, and when I push the Output to On, nothing happens.

The connector at the back, is firmly in its place. 

The power supplies DO work, just they both have to be in Local mode.

I am writing again the (almost the same) settings we have now:

Top:
Voltage: 13.25V
UV: 11.50V
OV: 15.00V
Power: 500W
Current: 20.00A

Bottom:
Voltage: 38.80V
UV: 24.00V
OV: 47.50V
Power: 1.5KW
Current: 60.00A

Since they work in Local mode, its not our highest priority at the moment, however it would be nice to solve it.

kt10aan, 2017-05-02:
We have been running the power supplies in Local mode, so I assume that issue should be closed for now.

closed 2017-05-02 by kt10aan