robotology/mesh-support #260: iCubSheffield broken cables

dcam0050, 2016-06-15:
##### Description of the failure
1. Torso yaw cable broken
2. Middle distal cable broken
3. Middle proximal cable broken

##### Detailed conditions when the failure occurred
Failure 1: Using it in normal operation turning to point to an object and saw the cable snap
Failures 2 and 3: Not present when they happened

Cables have been removed and would like to request a complete set of replacement cables. Thanks.

dcam0050, 2016-06-17:
Who do I need to contact to get a quote and order a full set of tendons? Thanks

julijenv, 2016-06-17:
Hi @dcam0050 ,
Do you have some spare cables, especially the torso yaw one (Number 13) to restore the torso in the meanwhile?
here below is an example I would recommend you to order to be safe for a long time.
It is a kit of at least 2 piece each and 5 for the most breakable and wearable cable (7 and 8).
let me know, what you think about it.
You could notice tendon V2 with a different code number - about this I'd like you to send me a couple picture a of one of your foream at the level of the wrist or without the forearm cover would be of a great help (there are 2 versions of it) to know if i dont give you too much cable to order :S.
thx in advance
[order.xlsx](https://github.com/robotology/icub-support/files/320884/order.xlsx)

dcam0050, 2016-06-20:
Hey @julijenv unfortunately we don't have any spare torso cables here to restore the torso so it would be great if we could ship the list of cables you suggest as fast as possible. 

Here are the iCub forearm pictures you requested:
[image]
[image]
[image]
[image]

Are these good enough to determine which version tendon v2 is?
Thanks

julijenv, 2016-06-20:
Hi @dcam0050 ,
everything is alright, the list I sent to you is perfectly fitting your need, I will send you an invoice (I mean one of my colleague which deal with the purchase office will send you an offer for the set of cable).
Could you please give your personnal mail address on skype to forward it to my colleague?
my skypeID : icub.support
thx in advance

dcam0050, 2016-06-28:
Any progress with the cables invoice? There's a deadline approaching soon so we need the cables urgently. Thanks.

closed 2016-07-04 by dcam0050