robotology/mesh-support #261: iCubSheffield request for upgrade + firmware update

dcam0050, 2016-06-15:
##### Description of the failure
No control of prosupination of either arm. Suggested course of action was upgrade of firmware and update to use yarprobotInterface instead of the robotInterface. 

Would like to request help in both regards.

pattacini, 2016-06-15:
The problem we saw on the pronosupination is the following: when put in compliant mode through `yarpmotorgui` we were unable to set both stiffness and damping coefficients of that joint. By contrast, changing those parameters for the preceding joints worked correctly.

@randaz81 suggested that this could be caused by old FW versions running on board.

julijenv, 2016-07-05:
any news about that issue?

dcam0050, 2016-07-05:
Issue is currently still there. I've installed the torso cable so the icub
is operational again. Thank you for that.

Now can we set up a date and time to update the iCub's firmware please? I
would like to request support this time because last time we had to ship it
over there to get it fixed after a failed firmware upgrade.

Also I do not know the procedure to upgrade from robotInterface which is
the one we are currently using to yarpRobotInterface.

Cheers,
Daniel

dcam0050, 2016-07-07:
Any availabilities?

julijenv, 2016-07-07:
Hi @dcam0050 ,
Asa far as I know your robot was recently shipped back to you and the firmware on the board has been updated at 99% certainty. You dont need to do any change of the firmware.
But for the change occured to the new way to laucnch the robot, you may have to download the updated icub-main repositry and recompile everything has ti should be explained on the wiki.
and afterwards everything should work as a charm.
Could you confirm that @randaz81 @pattacini ?

julijenv, 2016-07-07:
Hi @dcam0050 ,
I reread the issue and the firmware on the board 1B2 and 2B2 could have been older that the newest operateing on our robot? perhaps.
try to check with canLoader the version of the firmware of those boards.
let me know or join me on skype

pattacini, 2016-07-07:
Hi @julijenv 

I think in Sheffield they're still using `iCubInterface` instead of `yarprobotinterface`.

dcam0050, 2016-10-06:
Hey so I have just finished updating firmware and all software (<1 week old) and I still have the same issue with prosuppination. With impedence off, `ARE` performs OK but with impedance on there is no control of prosuppination. Could we arrange a skype call tomorrow to figure this out?

dcam0050, 2016-10-12:
Starting a separate issue for prosuppination

closed 2016-10-12 by dcam0050