robotology/mesh-support #263: Can't start icub server

fjandrad, 2016-06-24:
The server was off so we tried to turn it on but it never connected.

We tried to ping 10.0.0.1but only answer was host is unreachable.
We tried turning everything off and on 2 times but nothing changed.

fjandrad, 2016-06-24:
cc @mbrunettini  could you help us?

mbrunettini, 2016-06-27:
Boot issue.
Fixed.

pattacini, 2016-06-27:
cc @vtikha

traversaro, 2016-06-27:
@mbrunettini how do you solved the problem? It may be helpful to know for the next times it happens.

traversaro, 2016-06-29:
Apparently it happen again, after a mysterious automatic reboot while the server was running.

traversaro, 2016-06-29:
cc @DanielePucci @francesco-romano

mbrunettini, 2016-07-04:
The two issue was definitely NOT related:
1) misconfiguration on ESX bootup sequence after system upgrade.
2) ESX experienced a core dump (!!) error on June 29th, 16.31

The most recent issue was probably related to a memory issue (possible hardware fail?).
After that, the system took a lot to boot up due to filesystem checke and logs dump. Normally, the server boots up in less than 3 minutes (just checked).
Please keep the system under monitoring and let me know any anomaly.
Closing this issue and opening another one.

closed 2016-07-04 by mbrunettini

traversaro, 2016-07-04:
Thanks @mbrunettini .

claudiofantacci, 2016-07-04:
Hi guys, I've just *heard* that the server automatically reboot.
It actually happened twice.

cc @marcoaccame

mbrunettini, 2016-07-04:
1) What doeas it means HEARD?!?!
2) This issue is closed and please do not use it, see above.
THANKS

DanielePucci, 2016-07-04:
Refer to #265 for fishy server reboots.