robotology/mesh-support #264: Power Issues - iCubDaejeon01

mulkkyul, 2016-07-04:
Hi all,
We (iCubDaejeon01) have some issues with our power supply. The problems we have are:

- Once we shutdown the PC104 board, we cannot restart it immeadiately.
After we turn off the PC104 board (either by shutdown command or H/W green switch), we cannot restart PC104 immediately. 
It seems a sort of "cooling time" is required.. If we wait around 5 minutes, we can connect to the PC104 board successfully.
Is it normal to wait 5 minutes?

- OV occurs when the CPU/motor power switch is on.
(This problem seems solved, but I'm writing it for reference.)
We found that if we set our CPU voltage as described in Wiki (http://wiki.icub.org/wiki/Power_supply), OV occured when we turn on the CPU green switch.
We tested other PS settings and we could solve that problems by changing the CPU voltage to 13.25V.
We also experienced OV after we turned on the motor switch and that problem was solved by changing CPU OV setting to 15V.

Currently, our power supply settings are: 
(CPU) V: 13.25V, UV:11.5V, OV:15V, Power:0.5kW, Current:20A
(Motor) V:39.27V, UV:24V, OV:48V, P:1.5kW, C:15.3A
Would it be ok if our settings are different from Wiki?
(*We are using the ones from Elektro-automatik.)

[image]

julijenv, 2016-07-04:
Hi @mulkkyul ,
I know that those nice brandy power supply has for sure some delay (cooling) in between the switch OFF and ON, but it  appears that it is not as long as you mentionned, just take a look at the value for the voltage when you shut down the power, you should see the value going down, when it reaches a value close to zero it should be ready to repower up.
could you check this please?

mulkkyul, 2016-07-06:
Hi @julijenv 

Do I need to turn off the power supply to restart PC104?
What I've done was just switch on/off the CPU, NOT the power supply.

FYI, when I switch off the CPU, the current goes down to zero but not the voltage.
When I shutdown the power supply, the values go to zero and the PS is off. (it takes less than 10 seconds)

julijenv, 2016-07-12:
Hi @mulkkyul ,
You dont need to turn ON and OFF the power supply itself to be able to restart the pc104, just turn OFF the switch for the pc104 and wait 5s and reswitch back.
if the 12V is still ON on the front screen it is normal, the 12V is the main power supply which also give the authorization for the 48V power supply to turn on. the 12V is still ON because actually you still have 12V along the power cable but as long as the green switch for the pc104 is not turned ON therei= is no power pooring inside the robot ;).

mulkkyul, 2016-07-12:
Hi @julijenv 
Ok, then this problem might be related to our another issue about the PC104 boot failure (https://github.com/robotology/icub-support/issues/278). 
Thanks for the comment!

closed 2016-07-12 by mulkkyul