robotology/mesh-support #267: Neck tendons brokens

DanielePucci, 2016-07-04:
Please fill in the following two sections:

##### Description of the failure
The tendon of the neck is broken.

##### Detailed conditions when the failure occurred
While performing the YOGA++ demo, unexpected shaking phenomena occurred and the robot fell. 

**Remark**: if you are not a daily icub-support user (e.g. newbie), please read carefully the guidelines for contributing linked at the top of this page. In particular, the name of your robot must be always indicated: either using labels if you're already a contributor, as per standard, or by specifying it while detailing the issue.

@spada1978

DanielePucci, 2016-07-04:
@julijenv

spada1978, 2016-07-04:
replaced cable A number 5956

closed 2016-07-04 by spada1978