robotology/mesh-support #269: Various software issues on icub19 laptop

traversaro, 2016-07-06:
* There is no interned access on the laptop, even if DNS queries are correctly resolved and there is internet access on `icub-srv` . 
* `meld` is not running on the `pc104` (`meld` is called by `yarp-config robot --diff YARP_ROBOT_NAME`, so it is quite critical that it works correctly to main the robots configuration files). 

cc @mbrunettini

mbrunettini, 2016-07-06:
1) can you check that the IP address is 10.0.0.1 and the GW is 10.0.0.1?
2) did you export the DISPLAY variable before using MELD?

traversaro, 2016-07-06:
2) All other graphical programs are working, I can't easily copy paste the error message that only `meld` is outputting because the internet is not working : )

traversaro, 2016-07-06:
1) [image]

mbrunettini, 2016-07-06:
fixed.
1) wrong NATting configuration on server, due to reinstallation
2) locale not configured conrrectly on PC104, fixed in next image release

closed 2016-07-06 by mbrunettini

traversaro, 2016-07-06:
There is a workaround for 2) in the meanwhile? Setting `LC_ALL` in the terminal where you want to use `meld`?

mbrunettini, 2016-07-06:
it IS working on lumix now.
I fixed on the PC104, the next image will be working by default on alla images.