robotology/mesh-support #272: Broken right shoulder cable on icub black

tanismar, 2016-07-06:
##### Description of the failure
Broken right shoulder cable on iCub Black

##### Detailed conditions when the failure occurred
I was testing some tool use actions, and in one of the robot pulled his arm too close to the body, creating a tension that broke the cable :(

vtikha, 2016-07-06:
This is extremely unfortunate as the robot is needed by 9:00 tomorrow morning. Is it possible to have it fixed first thing in the morning? Thanks in advance

tanismar, 2016-07-06:
I know... I am very sorry... :( I hope it can get fixed on time :S

spada1978, 2016-07-07:
as your request repaired on time. cable number 7 replaced and the wiring again.

closed 2016-07-07 by spada1978

vtikha, 2016-07-07:
Thank you very much @spada1978