robotology/mesh-support #275: Some boards are unrechable in canLoader

mulkkyul, 2016-07-11:
**iCubDaejeon01**

##### Description of the failure
We're experiencing some problems while connecting to some boards in canLoader.

1. CFW2CAN2: Right arm
Board ID 13 (Strain) is missing in canLoader. Could you suggest a nice starting point to examine this issue? I don't know which cables/boards I should check.

1. CFW2CAN8: Skin Right Arm
Board ID 10 (2B10, RIGHT UPPER EXTERNAL) is missing in canLoader. The board seems fine, and both blue and green leds on 2B10 are on (please see a photo below). Is it okay if we replace 2B10 with 2B13 to check whether it's a board problem?

1. CFW2CAN5: Left hand
Additional Info of Board 14 shows some weird characters. Is it okay to have such message?

 
 

This is a part of our current problems mentioned in https://github.com/robotology/icub-support/issues/160. Some problems were (magically) fixed but still it's not clear how those problems were solved (So I left it opened)
Thanks!
 
 

##### Detailed conditions when the failure occurred
This is the picture of 2B10.
[image]

These are the screenshots from canLoader:
**NET 0**
[image]

**NET 1**
[image]

**NET 2**
[image]

**NET 3 & 4**
(Our poor iCub doesn't have legs, so I guess this error is okay.)
[image]
[image]

**NET 5**
[image]

**NET 6**
[image]

**NET 7**

[image]

**NET 8**
[image]

**NET 9**
[image]

julijenv, 2016-07-11:
Hi @mulkkyul ,
Ill try to answer in order to every single point:
1 - could you try to check the board 2B2? this board has a filter attached to a 4 pins header (filter = kind of protuberance - little board encpsulated in a transparent gummy cover). the label written on it should be N2. follow this N2 line until you reach the other filter BUT this one is cover with a black gummy cover. this filter is attached to the strain - Could you please check the continuity of the different wire and see if there isnt any wire broken.
2 - let's try to make some debug: answer to this question
remove 4 pin connector to 2 board (2B10 included) and swap them - can you see the board?
remove the 8 pin connector - can you see the board?
3 - the weird character are not a problem, please just reload the firmware MIS.hex and tick also the EEPROM case. this should remove the weird character.
let me know

mulkkyul, 2016-07-12:
Hi @julijenv , 
Thank you for the comments.

1. I checked the continuity of the wires and they seem fine. I checked the continuity of these 2 wires and there were no problems in continuity between the connectors in the red rectangles in the photo. 

[image]

2 & 3. I couldn't proceed to check these issues because we have unstable booting & CAN connection problems again.. (I though those problems were magically? gone, but sadly they still remain) I'm gonna open new issues related to these booting & CAN problems. Once I solve these problems, I'll get back to this issue and check 2 &3 as you suggested. Thanks!

julijenv, 2016-07-12:
Hi @mulkkyul ,
did you check the black filter? and the wires from the femelle connector which indedd is the input of the signal for the FT sensor (STRAIN)?.

mulkkyul, 2016-07-12:
Hi @julijenv ,
Thanks for the comments. Here are some more information.
- I checked the black filter and it has no discontinuity problem between the FT sensor to the 2B2 board. One thing is that there is a wire (black, white, green) connected to another side of the black gummy one. I checked the continuity between this wire and the one connected to back of the head (N2), and it seems they are disconnected. Since the other boards in NET2 were displayed in canLoader, I guess this might be okay?
- The transparent gummy cover has two connectors, but only one of them has wire. Could you check whether the wire is placed in the correct connector? I don't see any label on the gummy cover, so there's chance that I might have misplaced the connector. Thanks!

[image]

julijenv, 2016-07-12:
Hi @mulkkyul ,
could you check if in the geen box in the picture above the white cable which goes to the ti the end of the connector of the black filter, if there isnt a wire broken?
thx in advance

mulkkyul, 2016-07-13:
hi @julijenv ,
I check the wires and they seem fine. I also check the continuity from the yellow rectangle in the picture to the end of the connector which goes to 2B2 and they have no problem.

[image]

mulkkyul, 2016-07-13:
Here are some updates regarding the problem 2

**2. CFW2CAN8: Skin Right Arm**
- Swapping 4 pin connectors between 2B10 & 2B13 ==> Still 2B10 doesn't show up in canLoader
- Removing 8 pin connectors from both 2B10 and 2B13 ==> Still 2B10 doesn't show up in canLoader

[image]

julijenv, 2016-07-13:
Hi @mulkkyul ,
for the strain pblm;
try to remove the glue on the end of the wires (both side of the connectors), and check if you try to pull them out with a twiser if they snap or if they are firmly crimped.
if you cant find any wire broken or got snapped by the little push out try to recrimp all the wires (4 one side and 4 other side).
for the skin pblm:
could you try to unconnect the 8 pins connector on 2B10 and just connect another 4 pin connector on it and leave the other boards unconnected, see if on canLoader the board apperas?
let me now

mulkkyul, 2016-07-13:
Hi @julijenv ,

Regarding the skin problem, I tested several cases. Although I disconnected several other boards and connected 2B10 to another 4pin connector, 2B10 doesn't show up in canLoader. Here are some updates on this issue.

(8 pin connector to 2B10 board was removed)

2B10: Connected to the 4 pin connector which was connected to 2B13 board
2B13: Disconnected (Board ID 8)
[image]

2B10: Connected to the 4 pin connector which was connected to 2B13 board
2B13: Disconnected (Board ID 8)
+) 2B11: Disconnected (Board ID 11)
[image]

2B10: Connected to the 4 pin connector which was connected to 2B11 board
2B13: Disconnected (Board ID 8)
2B11: Disconnected (Board ID 11)

[image]

julijenv, 2016-07-13:
hi @mulkkyul ,
did some lights turn ON on the MTB 2B10?
Could you try to remove it and take a good look at it, oes it look like a component got burnt? try to resolder the connector on the MTB ( I mean just bring the tip of the soldering iron head with some soldering paste close to everysingle pins of the connectors to make sure that all of them or perfectly solder - maybe the soldering paste on the board crack by a shock due to reposition of the boards inside.
let me know

mulkkyul, 2016-07-13:
Hi @julijenv 

2B10 board seems fine, but it is assigned with different board ID. I found that 2B10 board has ID number 9, not 10 as written in wiki.  (http://wiki.icub.org/wiki/Can_addresses_and_associated_firmware#iCub_v2)

Here's are the screenshots.
When the 2B10 board was disconnected:
[image]

When the 2B10 board was connected:
[image]

julijenv, 2016-07-13:
Hi @mulkkyul ,
this what is fear at the beginning that why i ask you to unplug all the other skin boards (;D)
alright just change the ID for 10 and you ar done

mulkkyul, 2016-07-13:
Hi @julijenv, 

Thanks for the feedback! :) 
I changed the ID of the 2B10 board to 9. And I found that connection to 2B12 board have some problems. I don't see LED lights on 2B12. I'll debug it tomorrow and post the status.

Also, **the first issue (CAN2: Right arm - FT Sensor) is solved!** I don't know why it's working now, but disconnecting and connecting the connectors might help, I guess? 
[image]

mulkkyul, 2016-07-14:
Hi @julijenv ,

We confirmed that 2B10 board itslef has some problems. 
- No LED lights on
- the board is hot when the power is on (I guess this is a bad sign..)

So I check the continuity between the parts on the board. So far I found these problems:
- Several capacitors have continuity across the ends (C1, C4, C5, C7, C9 on the front side, C2, C3 on the back side)
[image]
[image]

- Pins in the yellow rectangle below seems connected each other. I check it with another board (2B10) and there was no continuity between these pins.
[image]

Any suggestions on this problem? and where can I find the schematic of MTB3_001 Board? 
Thanks!

mulkkyul, 2016-07-15:
Hi @julijenv ,

It seems that the 2B12 board is short-circuited (+3.3V - GND). I don't know what happened to it. :( 
Anyway, I'd like to make an order for the 2B12 board and a set of tendons, if possible. 
Who should I contact to get some (rough) estimation of the price and make an order? 
Thanks!

julijenv, 2016-07-18:
Hi @mulkkyul ,
I would need one picture of the entire icub, to explain to the guy who will create the draft for the quotation to know what version of the robot you have and sthe set of cable you need - it is a generical one. and one MTB, is it alright?

mulkkyul, 2016-07-18:
Hi @julijenv ,

Please see the pictures below. 
And yes, I wanna order one MTB_001 board and a set of spare items. Basically, we got nothing here in Daejeon.. so it'd be great if we can get some other spare items along with the board. Currently, I'm thinking about making an order for the following items
- 1 x MTB_001 board
- tendons for each in shoulder, hand, wrist, waist, and head (I guess having 3 for each tendon is enough?)
- 5m of steel spring tubes for the hand tendons
- 10m of teflon tube
- 10m of coated cables for the fingers
- 10m of uncoated cables for the fingers

Does this make sense? and if you can suggest some other items which might be needed in future, that'll be greatly appreciated.
Thanks!

[image]

[image]

julijenv, 2016-07-25:
Hi @mulkkyul ,
could you please send me a mail @ julien.jenvrin@iit.it in order to be able to interface you with the guy who will give you an estimated cost for the requested calbes and MTB.
thx in advance

julijenv, 2016-10-13:
hi @mulkkyul ,
any news from that matter?

closed 2019-10-12 by pattacini