robotology/mesh-support #278: PC104 Boot Failure

mulkkyul, 2016-07-12:
**Robot: iCubDaejeon01**

Hi all, we have some problems with booting the PC104 board. Although this issue was reported in my previous post (https://github.com/robotology/icub-support/issues/160), but I post this problem again to separate it from the other problems reported in that post. 

##### Description of the failure
When we turn on the robot (green CPU switch), boot fails (not always, but randomly). When the boot fails, there are 2 cases: either (1) PC104 doesn't start normally or (2) Linux bootloader (grub) is not loaded. (Please see below for detailed conditions)

The current solution so far was waiting 3 (or more) minutes and try again. This seems working sometimes but I think there must be better solutions.. I'm not even sure whether this is H/W or S/W problems. If anyone can share some tips, it'd be much appreciated.

##### Detailed conditions when the failure occurred
The monitor is connected to the PC104 board and we can observe the booting procedure. 2 types of boot failure were:

**1. PC104 doesn't start.** In this case, the monitor doesn't receive any signals from PC104. The monitor doesn't show anything on its screen. 

**2. USB is not recognized / grub is not loaded.** In this case, the monitor shows the BIOS screen during booting, but the booting hangs while showing a cursor blinking in the black screen (please see the video below). This seems that the linux bootloader (grub) is not loaded appropriately (or the USB is not recognized). We observed the same problem when we removed the USB stick from the PC104 board. VIDEO: [https://goo.gl/photos/bzzeWB1Huro65BSU6](https://goo.gl/photos/bzzeWB1Huro65BSU6)

mulkkyul, 2016-07-21:
Update on this issue:
- [x] Testing with another USB didn't work.

I burnt a new USB with the latest image (6.0) but it didn't work. Sometimes it still shows a blinking cursor on a black screen during boot.

julijenv, 2016-10-13:
Hi @mulkkyul ,
did you try to make sure that all the connectors are well plug into the pc104 board especially the one on which hang the usb stick?

closed 2019-10-12 by pattacini