robotology/mesh-support #279: PC104 freezes while opening any CAN devices

mulkkyul, 2016-07-12:
**robot: iCubDaejeon01**

##### Description of the failure

We're having problems while connecting to any CAN devices. Sometimes we could manage to connect to CAN devices, but in most cases PC104 freezes while we try to connect to any CAN devices. 

##### Detailed conditions when the failure occurred
from _dmesg_, we observed that CFW driver is successfully uploaded during booting (by the startup script - S02_cfw002.sh in /etc/rciCub.d).
```
[ 77.936611] Cfw002 device driver V 0.6.3
[ 77.936669] cfw002 0000:03:04.0: setting latency timer to 64
[ 77.936705] Cfw002 found!
[ 77.936709] Cfw002 registered with major 280, minor 0
```

Another startup script (S09_cfw2_sound.sh) is supposed to test the cfw002 driver during booting ([http://wiki.icub.org/wiki/CFW002_PC104_linux_driver#Startup_scripts]), but we commented it out since it freezes PC104 during the booting procedure. 

Once PC104 boots normally, we tested CAN connection by either (1) canLoader or (2) S09_cfw2_sound.sh script. When PC104 doesn't freeze while testing canLoader (or executing S09_cfw2_sound.sh), _dmesg_ shows followings:
```
[ 181.296026] 0000:03:04.0 (cfw002): Firmware cfw002_fw.bin needed
[ 181.306920] cfw002 0000:03:04.0: firmware: agent loaded cfw002_fw.bin into memory
[ 181.306927] 0000:03:04.0 (cfw002): uploading 30599 bytes firmware
[ 181.309724] 0000:03:04.0 (cfw002): image transfer done
[ 181.932031] 0000:03:04.0 (cfw002): Got firmware alive rensponse after 156 ticks
```

But in most cases, PC104 freezes. So I guess there are some problems while opening CAN devices.
I also removed all 10 cables (N0~N9) from the back of the head (CFW-002). But the problem still remains. When PC104 doesn't freeze, canLoader shows an error message saying "board not found". So I guess this problem is not related to CAN devices in the arms/torso.

To debug this issue, where should I start from? 
Any suggestions would be appreciated. Thanks.

julijenv, 2016-08-29:
Hi @mulkkyul ,
any news on this manner?

mulkkyul, 2016-09-01:
Hi @julijenv 
Unfortunately, this hasn't been solved yet :(
I'll get back to this issue later and give an update if we have any news.

julijenv, 2016-09-01:
Hi @mulkkyul , just to make sure:
have you try when the pc104 doesnt freeze to plug one CAN at a time and TURN ON the MOTORS switch in order to check the boards that should be detected one the specidfic line.
They are 2 ways for the CFW2 to send the mesage "Boards not Found", either the CAN line is broken (or maybe the component on the CFW2 board itself) or the boards are not powered.
try to test all the different CAN Lines until you got values inside canLoader (by connecting just one CAN  Line at a time and previously turn OFF the switch for the motors, and after you connected the next CAN Line, power the motors).
let me know your findings.
you can also give a call on skype. ID : icub.support

julijenv, 2016-10-13:
hi @mulkkyul ,
any news on this matter?

closed 2019-10-12 by pattacini