robotology/mesh-support #280: Issue with yarprobotinterface after updating firmware

dcam0050, 2016-07-12:
##### Description of the failure
The firmware was updated today and configured such that yarprobotinterface uses the xml files that come with icub-main instead of those in .local. The first issue was the log below:

```
[INFO]created device <canmotioncontrol>. See C++ class yarp::dev::CanBusMotionControl for documentation.
[INFO]/icub/head : no ROS initialization required 
[INFO]/icub/head  initting YARP initialization 
yarp: Port /icub/head/rpc:i active at tcp://10.0.0.2:10023
yarp: Port /icub/head/command:i active at tcp://10.0.0.2:10024
yarp: Port /icub/head/state:o active at tcp://10.0.0.2:10025
yarp: Port /icub/head/stateExt:o active at tcp://10.0.0.2:10026
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[DEBUG]POSITION_CONTROL: using metric_units
[DEBUG]POSITION_CONTROL: using control law joint_pid_v1
[WARNING]embObjMotionControl::fromConfig(): Warning: no VEL_PIDS group found in config file, skipping 
[DEBUG]TORQUE_CONTROL: using metric_units
[DEBUG]TORQUE_CONTROL: using control law motor_pid_with_friction_v1
[DEBUG]CURRENT_CONTROL: using metric_units
[DEBUG]CORRENT_CONTROL: using control law motor 2foc_feedback
[ERROR]kp  incorrect number of entries in BOARD IP = 10.0.1.5 
[ERROR]embObjMotionControl::fromConfig(): CURRENT_CONTROL section: error detected in parameters syntax 
[ERROR]Missing parameters in config file 
yarpdev: ***ERROR*** driver <embObjMotionControl> was found but could not open
[WARNING]Cannot open device torso_mc 
[WARNING]Cannot open device torso_mc 
```
For this error we went to the file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/hardware/motorControl and in the file torso-ems5-mc.xml, in the CURRENT_CONTROL section we noticed that there are 4 columns instead of 3.

We changed this to contain 3 columns and the error above does not happen but another issue replaces the error because when the icub starts up during the calibration phase, the icub torso joint rotates in the yaw axis to the left very fast.

Also we noticed another issue in the hands which are moving backward rather than forward in the calibration phase as well.

Finally, yarprobotinterface also shows this error: 
[ERROR]HasSpeedEncoder  parameter not found.

We tried using the xml files that were present in .local but using these files gives the following error in yarprobotinterface: 

[ERROR] from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 28s 450m 603u: (code 0x0500000c, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMAIS cannot be configured. it was already activated with different configuration. + .

Additional Information: This is the first time I am running yarprobotinterface because I was previously using robotInterface up to today. Also the torso yaw cable was changed last week.

Thank you for your help and for the help of everyone that aided in the discovery of these issues.

ale-git, 2016-07-12:
Yes, there were 4 columns in the CURRENT_CONTROL section instead of 3, sorry.
About the torso yaw, maybe the sensor has been remounted in reverse position? You should launch yarprobotinterface with external fault pressed, and then read the position value for torso yaw in the yarpmotorgui when the torso yaw is in 0 position, just to be sure that the position sensor is ok. 
About the HasSpeedEncoder parameter, it is strange because it is present in the configuration files. It is not mandatory anyway, even if the yarprobotinterface reports it as an error.

dcam0050, 2016-07-13:
I will check the torso yaw as you suggest.Thanks.

What about the hands moving backwards during the calibration phase?

marcoaccame, 2016-07-13:
Hi @dcam0050, as you may remember yesterday I renamed the folder `iCubSheffield01` under .local into `iCubSheffield01.tmp` so that yarprobotinterface did not use it.

Can you please post the entire folder in here?  

It contains the xml files which made the robot recently work,, thus it is worth comparing it with the folder which is now in git repository.

dcam0050, 2016-07-13:
Hey @marcoaccame when I was with @julijenv yesterday we tried using the .local files to launch yarprobotinterface but it generates these errors:

- `[ERROR] from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 68s 483m 493u: (code 0x0500000c, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMAIS cannot be configured. it was already activated with different configuration. + `
- `[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]  It has been detected that your control boards are not using the same
[ERROR]  CAN protocol used by iCubinterface. iCubInterface cannot continue.
[ERROR]  Please update your system (iCubInterface and/or your control board firmware.
[ERROR]  For further information please visit: http://wiki.icub.org/wiki/Firmware
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
`
- `[ERROR]cfw2can [0] board 1 CAN TX ERROR 
[ERROR]cfw2can [0] board 1 CAN TX OVERRUN 
[ERROR]cfw2can [0] board 1 CAN TX ERROR 
[ERROR]cfw2can [0] board 1 CAN TX OVERRUN 
Registering new thread 0 out of 500
[ERROR]cfw2can [0] board 1 CAN TX ERROR 
[ERROR]cfw2can [0] board 1 CAN TX OVERRUN 
[ERROR]cfw2can [0] thread:0 msg:91 joint:2 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[ERROR]cfw2can [0] thread:0 msg:91 joint:3 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[ERROR]cfw2can [0] thread:0 msg:91 joint:4 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[ERROR]cfw2can [0] thread:0 msg:91 joint:5 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version `

I think this is because the files in the .local worked for robotInterface but I had never used yarprobotinterface with those files. I've attached here a zip file with 2 folders. iCubSheffield01.localFiles with the .local files and iCubSheffield01.icubroot with the ones created after install for reference.

[iCubSheffield01.tar.gz](https://github.com/robotology/icub-support/files/361027/iCubSheffield01.tar.gz)

marcoaccame, 2016-07-13:
About the message printed when the old .local xml files are used:

```
[ERROR] from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 28s 450m 603u: (code 0x0500000c, 
par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMAIS cannot be configured. it was 
already activated with different configuration. + .
```
let me say that is it not relevant to the wrong movement of the robot. Also, the message appears only with xml files in the old .local folder which use an obsolete device.

The message just tells that the EMS has detected that two different devices are trying to initialise the MAIS CAN board with incoherent different settings. 

The guilty device is embObjAnalog, now superseded by the device embObjMais which you may remember I enabled yesterday in compilation phase.

The new xml files now use the embObjMais device which loads a MAIS configuration which is coherent with the FW of the boards.

marcoaccame, 2016-07-13:
OK, let me comment on your latest post.

1. `[ERROR] from BOARD 10.0.1.2`: see my previous post.
2. `[ERROR] [ERROR] It has been detected that`: it is a message from a CAN based device, hence it complains about FW protocol version of CAN board driving the head. Use canLoader on CFW2:CAN0 to update the FW w/ latest version.

About the attached files: thank you. It will be useful to do a comparison.

dcam0050, 2016-07-13:
@marcoaccame  We updated the firmware to the latest version yesterday for all CAN devices as well as for the ETH devices

dcam0050, 2016-07-13:
@ale-git I have launched yarprobotinterface with the new xml files and with the fault button pressed. IU then launched yarpmotorgui to check the torso values and the values make sense. I've attached a picture of the values together with one of the icub at those values

[image]
[image]

dcam0050, 2016-07-13:
Could someone tell me where I can find the angles for the home position of the icub so I can check that the encoders are correct for the arms? Right now the value of the shoulder pitch angle is -40deg in approx the usual home position.

dcam0050, 2016-07-13:
Also I've made these observations using yarpmotorgui:

- head:eyes_version, eyes are fully to the left at 0 degrees on the encoder

- torso immediately goes to the left fast when any torso joint is enabled

- right and left arms go backwards immediately when corresponding shoulder joint enabled.

- right and left elbow: No position controll when using slider. The right elbow allows me to open it readily by hand when it is in position control but stiff if I try to close the elbow joint. The opposite is true for the left elbow. It is stiff when I try to open it but readily closes by hand

- right and left wrist prosupination when enabled does not move but when using position control moves until reaches hardware stop and then goes into hardware fault. Position range between 1 and 2

- Joints 5 to 15 for right and left hand(wrist yaw, wrist pitch and all finger joints) show no position control and furthermore the position value range is between 1 and 2 in yarpmotorgui

dcam0050, 2016-07-13:
In short only the head moves correctly except for the eye version

dcam0050, 2016-07-13:
I have also noticed that if I disable the fault button a bit early when it is doing the hand calibration procedure, the thumbs, index and middle finger calibrate correctly but when then launched from yarpmotorgui they still have values between 1 and 2 possible for the movmement and cannot be controlled correctly.

dcam0050, 2016-07-13:
I changed the Position Control and Current Control values for kp to 10 and ki to 0 for the torso, left_upper and right_upper in iCubSheffield/hardware/motorControl in order to underpower them and allowed the yarprobotinterface to operate as usual with the fault button not pressed. This solved the issue with the eye_version and  the lower arms now operate correctly except for elbow control which I'm guessing is with the upper_arm group of joints. The faults are now limited to the torso joints, elbows and shoulders

ale-git, 2016-07-13:
@dcam0050 can you send me the config files in hardware/motorControl and hardware/mechanicals that were working fine, please?

dcam0050, 2016-07-13:
If you mean the ones that were in the .local folder which were being used by robotInterface, they are included in the iCubSheffield01.zip I attached above. I've included a link here as well

https://github.com/robotology/icub-support/files/361027/iCubSheffield01.tar.gz

dcam0050, 2016-07-13:
I've compared the files using a diff viewer and the only thing that has changed in the mechanicals between the .local files and the new xml files is the hasSpeedEncoder entry.

For the motorControl section the things that have changed are limited to the torso, left upper and right upper xml files (the joints which are right now faulty) and include two changes.
1) the addition of the current control section
2) the change of motor nominal currents and motor peak currents fropm 0 in the .local files to 5000 in the new files

dcam0050, 2016-07-13:
I think I may have solved the issue. I have changed the sign for the kp and ki values in the position control section of the cmls in hardware/motorControl and the icub is now at least booting up as usual now.

ale-git, 2016-07-14:
Hi @dcam0050, did you update the 2FOC firmware with the 2FOC.hex in icub-firmware-build two days ago? The version uploaded two days ago in the Git repository wasn't correct, sorry for the inconvenience... if you updated the 2FOC firmware with that version (now corrected in the repository) this could explain the need for Kp and Ki sign exchange and other issues.

dcam0050, 2016-07-14:
Yes we did a complete firmware update 2 days ago ... I will update again to the new firmware tomorrow and will keep you posted regarding issues.

julijenv, 2016-08-08:
any news about this?

dcam0050, 2016-09-16:
Hey everyone I've updated the CAN firmware of the icub for the foc boards on eth boards 1,3 and 5 but now I have a new error in yarprobotinterface.

`[INFO]EthResource::askBoardVersion() found BOARD EB3 @ IP 10.0.1.3 of type unknown with FW version = ( 2 , 48 ) and build date 2016 Jul 11 15:15 
[ERROR]EthResource::verifyEPprotocol() for ep = eoprot_endpoint_motioncontrol detected: pc104.version.minor = 17 and board.version.minor = 16 
[ERROR]EthResource::verifyEPprotocol() detected mismatching protocol version.minor in BOARD EB3 with IP 10.0.1.3  for eoprot_endpoint_motioncontrol : FW upgrade is required 
yarpdev: ***ERROR*** driver <embObjMotionControl> was found but could not open`

Which seems to be a mismatch between the expected protocol by the updated icub main and thee protocol present in the ETH boards. So I ran ethLoader and checked the version number of eUpdater which I expected to find at 2.16 but it is actually 2.17. Any ideas on how to progress from here?

dcam0050, 2016-09-16:
Screenshots for the ethLoader
[image]
[image]

marcoaccame, 2016-09-16:
Hi @dcam0050, you can proceed by reading https://github.com/robotology/QA/issues/135.

In case you need more help, let me know.

Also, subscribe to robotcub-hackers@lists.sourceforge.net to be alerted about changes or important posts in QA.

dcam0050, 2016-09-16:
Thanks ... I tried to do an update of the ETH boards but I found the instructions for updating the eth boards here https://github.com/robotology/icub-firmware-build/blob/master/ETH/howto-eth-boards-programming.txt do not contain enough information. The 2nd step of the process says to load the new ems file but it does not say which button is required to be pressed in order to load the file. Could you elaborate on this please?

marcoaccame, 2016-09-16:
To load the file you can use the button **Program eApplication**.

dcam0050, 2016-09-16:
Great! Thanks! The icub has started correctly now. Issue closed.

closed 2016-09-16 by dcam0050