robotology/mesh-support #282: Overcurrent on the right leg, hip pitch joint of iCubGenova05 (lumix)

pi-q, 2016-07-13:
##### Description of the failure

Starting up the iCubGenova05, in yarplogger the following error occured:
~~~
5869		2275,384728		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2452s 25m 496u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5882		2280,378629		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2457s 25m 491u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5914		2285,382293		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2462s 26m 488u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5928		2290,385744		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2467s 27m 491u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5933		2295,385956		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2472s 28m 491u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5946		2300,385213		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2477s 29m 491u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5959		2305,385599		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2482s 30m 488u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5972		2310,394088		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2487s 31m 491u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5985		2315,390568		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2492s 32m 491u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
5998		2320,388989		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2497s 33m 488u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
6004		2325,395525		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2502s 34m 488u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
6017		2330,391584		ERROR		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 2507s 35m 491u: (code 0x02000001, par16 0x0000 par64 0x0000000000000000) -> MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. par16 = ID of joint. + .
~~~

##### Detailed conditions when the failure occurred

It happened at startup. The failure can be reproduced using position control of the right hip pitch joint in yarpmotorgui.

traversaro, 2016-07-13:
Important note: this happends also if we set to 0 the Kp and Ki gains of the position control, and it also happens by just touching the leg without changing the position setpoint.

julijenv, 2016-07-18:
One of the board "FOC had one of the power compoennt with a pin unsoldered. risoldering this pin solved the issue

closed 2016-07-18 by julijenv