robotology/mesh-support #283: Mechanical Backlash on the right leg

S-Dafarra, 2016-07-15:
##### Description of the failure
When trying to set the hip roll joint to the home position, the right leg starts oscillating, especially when the error from the reference gets close to zero.

It happens all the time the robot is set to the HOME position and it has to be killed.

@julijenv  @francesco-romano

pattacini, 2016-07-15:
@S-Dafarra you should have received an email inviting you to join our dev group in robotology. This way you'll be entitled to assign labels to issues.

julijenv, 2016-07-18:
the screws holding the internal harmonic elipsoidal shape with teeth were all totaly loose. . It has been performed on one of those screws some unscrewing glue and all the other have been tightened.

closed 2016-07-18 by julijenv