robotology/mesh-support #286: Electronic board on the chest got damaged

pattacini, 2016-08-21:
Hi @robotology/icub-support-team 

I've received an email from our partners in Grenoble reporting the following major failure on their iCub.

##### Description of the failure

After an intense session of work - which lasted the whole day - they've seen a "_flash and smoke coming out of the iCub torso_". They suspect a failure of the electronic boards located in the iCub breast.

##### Detailed conditions when the failure occurred

On Thursday Gerard Bailly and Duc-Canh were conducting experiments to gather visuomator calibration data for pointing gestures. They started at 10:00 with no interruption until 18:00 they got the failure.

@julijenv please get in touch with them as soon as you can.

cc @EliseiFrederic @Fabrizio69

EliseiFrederic, 2016-08-22:
Hi,

I was not with them when it arrived, but an important information that Gerard forgot to report is that **the arms just fall (and not the rest of the robot) when it flashed.**
They did not think to look at the RobotInterface real-time log to spot new errors.

We´ll disassemble the torso cover as soon as the people from the technical staff are available.

julijenv, 2016-08-22:
Hi @EliseiFrederic ,
I briefly read what happen to your robot, and i'd like you to check the integrity of the boards at the level of the torso.
First of all, could you remove the torso cover (remove on both side of the skin cover 3 screws starting from the bottom , at that point you will be able to flip the skin and remove the 2 screws at the level of the chest and at the end the 2 screws at the level of the shoulder) and disconnect the wire on the cover skin.
From that point you will be able to inspect if the boards below the chest cover got damaged, (burnt components or loosing screws which could have finished inside the boards, etc...) please do some pictures of the boards you can identify.
If no third part components are hanging inside the boards, please turn ON the motors switch and check with canLoader what it can detect on the net0. Take a picture or list the different boards.
Let us know,

pattacini, 2016-08-22:
cc @NguyenDucCanh

EliseiFrederic, 2016-08-22:
Thank you Julien for the dismantling instructions. No lossed part. Here is what we found on the PCB : 

[image]
[image]
[image]

Our local Julien identified an explosed tantalum capaticor. See the attached images for the position.
Some "grease" is spread around a little, but nothing else looks damaged.

What is the next step ? Replacing the card ?

Thanks for your fast support.

julijenv, 2016-08-22:
Hi @EliseiFrederic  and @NguyenDucCanh ,
the first thing is to make sure the 2 boards are still ok (usually this tantalium got burnt whenever the voltage go over a certain value (in this case 16V) or an overcurrent). The board is usually still working if it is just the capacitor which got burnt.
try to replace it and before reconnect everything, just check that the outputs ports (P1, P2, P3 and P4) are giving you the following pin to pin voltage (12V,12V,12V,0V,0V,0V,5V,3.3V). And of course check the leds turns ON (DL1, DL2).
Check if ever the connectors have a no broken wires or fused wires from one side or the other (the L3, L4 and L5) goes up to the upper arm and power the MC4 borads(1B2, 1B3 and 1B4) check the integrity of the connector and boards there. (continuity test or diode test), in a symmetrical way proceed with R3,R4 and R5 with the boards located into the right upper arm cover. 
At that point reconnect the 3 connectors (L3, L4 and L5 or R3, R4 or R5) I can't really see the labelling on the pictures :S.
Check also the outputs for the other MCP boards the one which got blacked by the fumes of C3.
LEt me know your findings. or get in touch with me on skype: icub.support

EliseiFrederic, 2016-08-22:
Thanks, I forwarded your instructions to Julien Minet.

What is the reference of the C3 capacitor ? Should it be replaced by the same reference or a stronger one ?

julijenv, 2016-08-22:
same one as the one next to it on a symmetrical horizontal plan (which means C2).
[image]
hope it will help

EliseiFrederic, 2016-08-23:
Could it be this ?
I'm unsure about the case/size of the SMT :-(
http://fr.farnell.com/kemet/t498x107k016ate075/tantalum-capacitor-100uf-16v-case/dp/2343389

EliseiFrederic, 2016-08-23:
At some places in 
https://svn.robotology.eu/repos/iCubHardware-pub/trunk/electronics/public/MC4-PLUS/
I found that this C3 is "IIT reference 1035"
Were can we found this partlist ?

Google suggested
https://svn.robotology.eu/repos/iCubHardware-pub/trunk/electronics/public/BCB/Partlist/bom_tr_6117_A.csv
but the entries for 1035 look quite different to what we see on our PCB.

Am I using outdated files ? Links from http://wiki.icub.org/wiki/Manual#One._Hardware_of_the_iCub to electronics sometimes go to source (code) repository :-(
For example the "bills" links from http://wiki.icub.org/wiki/Other_boards

julijenv, 2016-08-23:
Hi @EliseiFrederic ,
you went into MC4-PLUS but you should have check the repository MCX-001 and inside you have both MCP and MC4 board, in your case the MCP board (power board)
C3 is this component from BOM
farnell code : 4765680 or simile
[image]

good luck

pattacini, 2016-08-24:
cc @baillyge

julijenv, 2016-08-26:
Any news?
@EliseiFrederic , @NguyenDucCanh

EliseiFrederic, 2016-08-29:
Hi, our local Julien has ordered the capacitors. He'll be back Wednestay to solder them !

EliseiFrederic, 2016-09-01:
Julien did the repair, with a new capacitor. Robot seems to behave and move normaly again.
Thank you for the support !

closed 2016-09-01 by EliseiFrederic