robotology/mesh-support #287: Errors on starting robotInterface

dogoepp, 2016-08-24:
When launching `robotInterface`, we regularly get errors with either the right arm or the left leg. I attach the log files in [a zip archive](https://github.com/robotology/icub-support/files/434382/log-files.zip) for two such situations.

Bellow is an excerpt of `ypmrprunlog_03_08_2016_05_31_46.log` highlighting the error messages that seemed most relevant to me.

> from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 171s 52m 291u: (code 0x06000002, par16 0x0002 par64 0x0000000000000001) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = P2, 1 = P3, 2 = internal). Number of errors in par64 + .qq  
> unable to complete calibration: joint 4 in 'hw_fault status' inside doneRaw() function  
> Right_Arm_Calibrator : Timeout while calibrating  8  
> unable to complete calibration: joint 5 in 'hw_fault status' inside doneRaw() function  
> Right_Arm_Calibrator : Timeout while calibrating  9  

It keeps going with this kind of messages for joints 11 and 13 (timeout for right arm calibration) and 7 and 9 (unable to complete calibration)

Later in the log, we have a series of new messages:

> Right_Arm_Calibrator : set 4 : Calibration went wrong! Disabling axes and keeping safe pid limit  
> embObjMotionControl::checkRemoteControlModeStatus(4, 7103593) is unable to check the control mode of BOARD EB4 IP 10.0.1.4 because it is now in HW_FAULT  
> embObjMotionControl::checkRemoteControlModeStatus(5, 7103593) is unable to check the control mode of BOARD EB4 IP 10.0.1.4 because it is now in HW_FAULT  
> embObjMotionControl::checkRemoteControlModeStatus(7, 7103593) is unable to check the control mode of BOARD EB4 IP 10.0.1.4 because it is now in HW_FAULT  
> embObjMotionControl::checkRemoteControlModeStatus(9, 7103593) is unable to check the control mode of BOARD EB4 IP 10.0.1.4 because it is now in HW_FAULT  
> from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 286s 786m 750u: (code 0x00000013, par16 0x0002 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .  

In that case, we stop the robot interface, power cycle the motors and try again. These errors appear regularly but we did not detect any plausible cause.

Do you have an idea of plausible causes and how to address them ?

By the way (it might not be related at all), the left leg keeps oscillating aroung its target pose.

Thanks a lot

cc: @jbmouret @serena-ivaldi @olivierrochel-inria

julijenv, 2016-08-26:
Hi @dogoepp ,

Does the robt calibrate? I dont really understand what is really the visible problem?
CAn you see the MAIS board attach to EMS 10.0.01.4? Can you see the MAIS lights at the level of the knuckles?
can we get in touch by skype. our ID is icub.support, lokk for us

cc: @jbmouret @serena-ivaldi @olivierrochel-inria

dogoepp, 2016-11-10:
Hi Julien,

Sorry for not answering before. Quite a few other things happened on the robot and beside it.

We don't get this specific problem any more. Hopefully it was solved by the software and firmware updates made in the meantime.

Thanks for your help.

closed 2016-11-10 by dogoepp