robotology/mesh-support #288: Boards non responding to `pc104` on iCubLyon01

gregoire-pointeau, 2016-08-25:
### Description of the failure
Hello, we have a failure when launching `yarproborinterface`.
It seems that the boards don't respond to `pc104`
It comes after a full update of the system.

### Detailed conditions when the failure occurred
Here is the log of the error we have: 

>AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
sharedcan [1] thread:0 msg:91 joint:0 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
sharedcan [1] thread:0 msg:91 joint:1 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
sharedcan [1] thread:0 msg:91 joint:2 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
###################################################################################
###################################################################################
  It has been detected that your control boards are not using the same
  CAN protocol used by iCubinterface. iCubInterface cannot continue.
  Please update your system (iCubInterface and/or your control board firmware.
  For further information please visit: http://wiki.icub.org/wiki/Firmware
###################################################################################
###################################################################################
checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
sharedcan [5] thread:0 msg:91 joint:0 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
sharedcan [5] thread:0 msg:91 joint:1 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
###################################################################################
###################################################################################
  It has been detected that your control boards are not using the same
  CAN protocol used by iCubinterface. iCubInterface cannot continue.
  Please update your system (iCubInterface and/or your control board firmware.
  For further information please visit: http://wiki.icub.org/wiki/Firmware
###################################################################################
###################################################################################
checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
lAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
lAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
One or more devices failed opening... see previous log messages for more info 

Thanks for your help.

julijenv, 2016-08-25:
hI @gregoire-pointeau ,
could you please do a canLoader and check the ID and version of all the boards for any net?
youu could send me the pictures of all the net, to cjheck with you if you wish or joining me on skype,
c u around,

gregoire-pointeau, 2016-08-25:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

julijenv, 2016-08-25:
hi @gregoire-pointeau ,
did u update the firmware on the boards? did you turn ON the motors?
let me know

gregoire-pointeau, 2016-08-25:
Latest error:

>AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
lAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
lAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
Head_Calibrator : invalid number of startupPosition params 
Torso_Calibrator : invalid number of startupPosition params 
Left_Arm_Calibrator : invalid number of startupPosition params 
Right_Arm_Calibrator : invalid number of startupPosition params 
Left_Hand_Calibrator : invalid number of startupPosition params 
Right_Hand_Calibrator : invalid number of startupPosition params 
One or more devices failed opening... see previous log messages for more info

gregoire-pointeau, 2016-08-25:
Full log is:

[log_25_08_2016.txt](https://github.com/robotology/icub-support/files/437050/log_25_08_2016.txt)

pattacini, 2016-08-25:
@gregoire-pointeau try to use labels to identify the robot instead of specifying the robot name within the title of the issue :wink: 

Here are the [**guidelines**](https://github.com/robotology/icub-support/blob/master/.github/CONTRIBUTING.md) to post to `icub-support`.

gregoire-pointeau, 2016-08-25:
After @julijenv update, new log: 
[log_25_08_2016.txt](https://github.com/robotology/icub-support/files/437360/log_25_08_2016.txt)

@pattacini I'll do ! (but in between icub-main, and icub-support I'm not always allowed to change labels, this is why I didn't do it here. But now, I know, don't worry !)

pattacini, 2016-08-25:
@gregoire-pointeau yeah, users can **assign labels** only in [`icub-support`](https://github.com/robotology/icub-support) and [`Q&A`](https://github.com/robotology/QA), because  `write` permissions is required to do that and we tend to be not that _generous_ with `yarp` and `icub-main` repos :wink:

julijenv, 2016-08-26:
After a complete update, always try to laucnh the robot from the main repostirory (in that case @randaz81  made some modifications in your config files which didnt allow you to start with your old config files).

gregoire-pointeau, 2016-08-29:
Robot starts and all joints respond.
However the calibration is not perfect (the arms aren't aligned very well).

[log_29_august_12h45.txt](https://github.com/robotology/icub-support/files/442232/log_29_august_12h45.txt)

gregoire-pointeau, 2016-08-29:
After a fine calibration, everything went back to normal.
Thanks !

closed 2016-08-29 by gregoire-pointeau