robotology/mesh-support #290: Right leg FT broken (EB8)

hu-yue, 2016-08-29:
### Description of the failure

I suspect that the right upper leg FT sensor (EB8) is broken.

### Detailed conditions and logs of the failure

The robot fell, and then it started complaining about FT sensors (all of them) saturating even if I lifted the robot from the floor.
I restarted the robot and then only the one on EB8 (right upper leg) had an error on saturation, even if the robot is hanging in the air.
Log [here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/ypmrprunlog_29_08_2016_06_17_17.log).
Reading from the port also gives only some constant values:
```
yarp read ... /icub/right_leg/analog:o
yarp: Port /tmp/port/1 active at tcp://10.0.0.100:10142
yarp: Receiving input from /icub/right_leg/analog:o to /tmp/port/1 using tcp
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
-1440.0 -1637.0 -1932.0 -31.0 -33.0 -22.0
```

hu-yue, 2016-08-29:
cc @KevinAStein FYI

hu-yue, 2016-08-30:
@robotology/icub-support-team
I checked the cable connecting the FT sensor to the board and it seems fine.
What else can I check/do?
Thanks!

marcoaccame, 2016-08-30:
Hi @hu-yue,

I don't know what the actual problem of the STRAIN boards can be, but surely @julijenv or someone else who will take care of the issue will help you solve.

My contribution can be to help understanding the log file. 

The log file says:

1. All the STRAIN boards are found and have the latest FW (application v 3.0.5, can prot v 1.0).
2. The STRAIN at 10.0.1.8:CAN2:13 saturates in high range of channel 4 for every single reading it does (its rate is 10 ms).
3. There is no other detected problem in the other STRAIN boards.

Here is evidence of the above.

The correct discovery is told by the following 4 messages.

```
<#STRING_START#> from BOARD 10.0.1.6 (EB6) @ 27s 237m 444u: CAN discovery has detected 
a eobrd_strain board in CAN2 addr 13 with can protocol ver 1.0 and application ver 3.0.5 
Search time was 0 ms
<#STRING_END#>
```

```
<#STRING_START#> from BOARD 10.0.1.8 (EB8) @ 28s 324m 738u: CAN discovery has detected 
a eobrd_strain board in CAN2 addr 13 with can protocol ver 1.0 and application ver 3.0.5 
Search time was 0 ms
<#STRING_END#>
```

```
<#STRING_START#> from BOARD 10.0.1.7 (EB7) @ 29s 404m 656u: CAN discovery has detected 
a eobrd_strain board in CAN2 addr 1 with can protocol ver 1.0 and application ver 3.0.5 
Search time was 0 ms
<#STRING_END#>
```

```
<#STRING_START#> from BOARD 10.0.1.9 (EB9) @ 30s 493m 59u: CAN discovery has detected
 a eobrd_strain board in CAN2 addr 1 with can protocol ver 1.0 and application ver 3.0.5 
Search time was 0 ms
<#STRING_END#>
```

The fact that the STRAIN at 10.0.1.8:CAN2:13 saturates comes from the following message which repeats every 10 seconds. 

```
<#STRING_START#> from BOARD 10.0.1.8 (EB8), src CAN2, adr 13, time 39s 354m 232u: 
(code 0x01000002, par16 0x0004 par64 0x000003e800000000) -> HW: strain values saturation.
In p16 there is the channel involved, in p64 there are the saturation counts for the last sec 
of both types (lower: LS32bits, upper MS32bits + .
<#STRING_END#>
```

hu-yue, 2016-08-30:
@marcoaccame thanks for the explanation, I have realised that the other FTs are probably fine.
I will wait @julijenv or anyone else in charge for further investigation instructions as now I am not sure what to do.

julijenv, 2016-08-30:
Hi @hu-yue ,
From the log phrases sent by yarprobotInterface, it seems that just the FT from the EB8 has an issue - saturation values NOT all the FTs.
The bad thing is that all the FT sensors respond so it means that the CAN line is not broken (the only fixable part of the FT sensor) and by so I mean that unfortunately I fear that the FT sensor is broken inside it.

hu-yue, 2016-08-30:
@julijenv thanks, that is also what I am afraid of.
How can I check if it is actually broken? 
Should I unmount it? If yes, how? (Does this involve unmounting the whole leg because it's in between the hip and the rest of the leg?)
I am leaving for Genoa tomorrow, should I unmount it and bring it to IIT?

julijenv, 2016-08-30:
@hu-yue ,
remove the cover of he upper leg
and after remove the 8 following screws
[image]
and after the 4 following screws,
[image]
from that point yu will have that part of piece in your hand:
[image]
at that point just remove the upper screws and the lower screws:
[image]
[image]
now you have the sensor in your hand. But just remember that nothing tells me that we can do anything to make working again, you will maybe need to buy another one, we will give an eye to it, but needless to say that I dont know if can give you one,
good luck,

hu-yue, 2016-08-30:
@julijenv thanks for the instructions, I will try.
Few more questions, if I bring it to IIT this Thursday or Friday (1st or 2nd Sept), how long do you need to check it? I will come back to Heidelberg on the 10th and we have some demos to show on the 14th (that might not involve the FT sensor), so would you be able to check it before the 10th? 
And in case I can bring the broken one back to use as placeholder in case you don't have a new one?

hu-yue, 2016-08-30:
@julijenv I successfully detached the sensor from the right upper leg.

I will see you in IIT then.
Thanks!

[image]
[image]

traversaro, 2016-09-02:
[image]

traversaro, 2016-09-02:
The `matrix_SN184.txt` stored in https://svn.icub.iit.local/repos/mecha/ftSensCalib/trunk/ contains:
~~~
FEFB
1
3D1D
276
149
C1E5
FDDF
3D27
E084
DE
FF12
E0A2
2A17
124
FDBA
284B
2934
FF09
FDFE
19A5
F423
D8F0
23AD
F297
DAC5
FF28
EA00
16EC
164A
14A7
8D
2836
26E5
FE31
FFCC
2946
1
1440
1637
1932
31
33
22
~~~

traversaro, 2016-09-02:
Wrap up: 
Apparently the calibration matrix in the eeprom is sane. 

The sensor is always overflowing on raw channel 4 (even at startup) and that is the reason why we are getting those strange measurements value: on overflow the software is supposed to send the last valid measurements, but if no valid measurement was ever received it seems that it is sending the fullrange of the raw channels. ?_? 

We changed the DAC offset (as you can seen in the following image) and at the moment the FT sensor seems to be working "fine" (no quantitative test was done, just a quick qualitative test on the workbench). We (with @DanielePucci ) to try at least once on a real robot experiment before sending it back to Heidelberg next friday.

traversaro, 2016-09-02:
New offset: 
[image]

hu-yue, 2016-09-11:
The sensor has been brought back to Heidelberg and remounted on iCubHeidelberg.
With the torque balancing demo with contacts switching on both sides, there were some saturation on channel 3 of the FT on EB8, on channel 2 of EB6 and on channel 0 of EB9, by moving the offsets of these channels it did not complain about saturations anymore for the whole duration of the demo.

This however does not ensure that the sensor(s) is(are) 100% all right, but for the time being this is ok for us, until we encounter further problems.

Closing for now, reopening in case of future problems.

hu-yue, 2016-09-14:
I am reopening this issue.

Problems encountered during the demo preparation for the KoroiBot workshop are as follows:
- FT on EB8 (right upper leg) saturates during the balancing demo, the offset has been moved, then it stops saturating for some run, then some time later (or the day after) it saturates again, and this happened for several times. I don't think moving the offsets forever is a solution, also because after I've moved it to one side, it started saturating on the other side, and I would not keeping moving forth and back the offset. This happens mostly on channels 3 and 4. This has caused the robot to fall during the actual demo, while it worked three times in a row one hour before.
- Not 100% sure that this is due to the sensor, but it's definitely happening on the same part (right upper leg). During balancing, a high spike of torque was generated and right after it caused encoder spikes on the first 4 joints (those controlled by EB8), which makes the robot fall immediately. Then the encoder of some of these joints stay on very high numbers (like 300-400 degrees) and do not go back to 0 unless the robot is switched off and on again. This happened at least two times on both right hip pitch and right knee.
Warnings are as follows:
```
4687 2986.668799 <WARNING>  from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 168s 186m 170u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000015) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
 
4688 2986.668940 <WARNING>  from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 168s 186m 192u: (code 0x0200000a, par16 0x0001 par64 0x000000000000001e) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
 
4689 2986.669018 <WARNING>  from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 168s 186m 212u: (code 0x0200000a, par16 0x0002 par64 0x0000000000000044) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
 
4690 2986.669088 <WARNING>  from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 168s 186m 232u: (code 0x0200000a, par16 0x0003 par64 0x000000000000004a) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
```

What should we do? 
1. Unmount the sensor and send it back to IIT again?
2. Get a new sensor?

For case 1. if it takes long to check it, we would need a dummy in order to use the robot for other purposes (e.g. position controlled walking or so). In any case we don't have too much time, we would need the robot to make experiments before the KoroiBot final review which takes place in November, and maybe also for the review itself. So we would be glad if this can be solved as soon as possible.

Thanks!

@iron76

iron76, 2016-09-19:
@hu-yue 
I would say that the saturation problem is well known in IIT and I would talk with @DanielePucci before being sure that the sensor is broken and needs replacement. Getting rid of the sensor saturation is a long standing activity and we don't have a clean solution. As far as I know we @ IIT play with offsets quite a lot but I let @DanielePucci confirm this. 

Anyway, we will send you a new sensor so that you have a way to compare the two. Please be aware that we might have tricks you are not aware of (e.g. there is a 30 min warming up phase and during this phase saturations would change quite a lot). @DanielePucci can help you with these tricks.

hu-yue, 2016-09-19:
@iron76 Ok thanks, I can do further tests with the insights that @DanielePucci can give me while waiting for the new one.

DanielePucci, 2016-09-26:
Guys, deeeeeeply sorry for taking so long to reply. Anyways, as @iron76 mentioned, the warming up effect of FT sensors is usually dealt with by performing frequent calibrations of FT offsets through WBDT. As a rule of thumb, once every 10 minutes should be fine. @hu-yue let me know if the problem still stands there.

julijenv, 2016-10-04:
Can we close??? @hu-yue

hu-yue, 2016-10-04:
@DanielePucci I just saw your answer now, thanks for the tip, we'll keep that in mind!
@julijenv I guess we could close it for now. The new sensor has arrived and has been mounted on the robot, we don't know if the old one is actually broken, but the basicl problem is solved.

closed 2016-10-04 by hu-yue