robotology/mesh-support #292: Contact failure with the ethernet câble

AnneLaureM, 2016-09-09:
Hello iCub-Support

A few months ago, we had some contact failures with the Ethernet cable of the iCub, so we have improvised a fix with our available material at INSERM and we have changed the connector on the head of the iCub.
[image]

But we think it will be better to repair correctly the connector.
Do you think it will be possible to send us a new connector and some crimps ?

Cheers,

Anne-Laure

julijenv, 2016-10-04:
feedback about that? can I close?
@gregoire-pointeau , @AnneLaureM

AnneLaureM, 2016-10-05:
We have received the connector. Thanks !

closed 2016-10-05 by AnneLaureM