robotology/mesh-support #293: iCub power supply not working

AlexAntn, 2016-09-09:
iCub power supply is no longer working, for iCubPlymouth01

Upon turning on the iCub power supply, after summer break, we heard a snap. Upon closer inspection of the power supply, we noticed the lower box was not presenting any voltage numbers (was, effectively, off).

Checking the fuses on the power plugs, both fuses are ok.

vtikha, 2016-09-16:
@AlexAntn, sorry for the late reply. Could you please send us some pictures (front and back) of the power supplies you have, in particular the lower box? We will try and determine a few things and take it up from there.

AlexAntn, 2016-09-16:
Update:

Today we opened the power supply unit, to check for any blown fuse. We found the fuse on the front-right to be blown, so we replaced it and turned it on again.

Upon turning on, the new fuse was also blown, so we assume there is some short-circuit on the PSU itself.

How can we replace it? 

Details on the current non-working PSU:
XFR-2.8Kw-60-46, Xantrex, - Power supply, 2,8 Kw, 0-60V, 0-46A, 89x430x534mm, 15kg, 19" rack mount, 2U
Order code: XFR-2.8Kw-60-46

AlexAntn, 2016-09-16:
Pictures of the Power Supply Unit:
(Lower unit is the broken one)

Front:
[image]

Rear:
[image]

Detail on top:
[image]

Detail on front:
[image]

Thanks @vtikha !!

vtikha, 2016-09-16:
@AlexAntn thanks for the update and pics, we will get back to you asap.

vtikha, 2016-09-16:
@AlexAntn unfortunately it is busted 👎 . I sent you an email containing all the details necessary for you to proceed.

closed 2016-09-16 by vtikha