robotology/mesh-support #296: Cannot boot pc104 / pc104 unreachble

hu-yue, 2016-09-19:
### Description of the failure
After switching on pc104 it is not possible to reach it. Ping fails and the pc104 does not appear in the network.

### Detailed conditions and logs of the failure
We don't know how this happened, it was all right when we used it last time on Friday, then today we switched it on and pc104 couldn't be reached. We don't know how to check if it's a problem of the pc104 not booting at all or other problems.
We did the following until now:
* made sure that the router is fine, all the other computers plugged in can see each other
* unplug the extension cable to make sure it's not there the problem, and still couldn't see the network
* tried to ping pc104 and it is unreachable, tried to ping the boards, in this case nothing happens, it does not even say unreachable
* plug ethernet cable of the robot directly to the laptop and used `ethtool` to check if the cable is fine, the result we got is the following:
```
ethtool eth0
Settings for eth0:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Speed: 1000Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 2
	Transceiver: internal
	Auto-negotiation: on
	MDI-X: on (auto)
Cannot get wake-on-lan settings: Operation not permitted
	Current message level: 0x00000007 (7)
			       drv probe link
	Link detected: yes
```

The question is: how do we identify where the problem is?
Thanks!!!

hu-yue, 2016-09-19:
For some reasons, now it works again.
I unmounted the usb stick from the robot, plugged it into another pc and tried to boot, it booted without issues. Then I replugged it back to the robot, and it started working again. Relaunched for three times and the problem did not occur again.
I will close the issue for the time being, until this problem shows up again, as I cannot quantify why it happened and why it stopped happening.

closed 2016-09-19 by hu-yue

barbalberto, 2016-09-20:
Sometimes, when pc104 is powered off with the hardware switch without shutting it down with "sudo poweroff", it may happen that the filesystem get stuck in a non clean state.
Plugging the usb in a pc and running a fsck, will cleanup the state of the filesystem and allow the pc to start again.

Latest version of the image is a live system, it should reduce, if not avoid, this problem.

gsaponaro, 2016-09-20:
Hi @barbalberto, short question: do you recommend using "sudo poweroff" or "sudo halt" on the iCub PC104? I was just reading about the differences in http://unix.stackexchange.com/questions/205464/whats-the-difference-between-poweroff-and-halt

barbalberto, 2016-09-21:
For the pc104 I think there is no difference since the power supply is external, and the green switch has to be turned off manually in any case.
However I usually go with `sudo poweroff` or `sudo shutdown -h now`