robotology/mesh-support #298: broken cable at the shoulder

reafrancesco, 2016-09-26:
Please **fill in the following two sections** and then **remove any comment and/or remark become unnecessary**:

### Description of the failure

### Detailed conditions and logs of the failure

---
**Remark**: if you are not a daily icub-support user, please read carefully the guidelines for contributing linked at the top of this page. In particular, the **name of your robot** must be always indicated: either using **labels** if you're already a contributor, as per standard, or by specifying it while detailing the issue.

julijenv, 2016-09-26:
Hi @reafrancesco ,
Right or Left arm? which cable are u talking about? Do u need us to put back in place the shoulder cables or can u do it by yourselves?
Could you try to sum up in a better way or give more details, thx in advance.
Now I'll try to see with Giorgio what are the administrative details we need to do for going to Erzelli.
Try in the meanwhile to answer to my specific question aboce mentionned,
let us know,
C u later
thx

reafrancesco, 2016-09-28:
The arm is the left. The cable at the elbow broke after the robot hit the table. I assume it is possible to remove the arm and bring it to Morego.

julijenv, 2016-09-30:
The cable N10 was broken and has been taken by Rea to restore is iCub.

closed 2016-09-30 by julijenv