robotology/mesh-support #299: Simple Application

appkof, 2016-09-26:
**IcubNTU**

Simple walk through application to start working with the Icub.

I have just started going through the documentation on www.icub.org, do you have a "Hello World" kind of application that I can start with.
Also I wanted to know if it's possible to connect a monitor (via VGA or similar) to the Icub.

lornat75, 2016-10-04:
Hi icubNUT,
there is some material here:

http://wiki.icub.org/brain/icub_tutorials.html

It is possible to attach a VGA cable to the iCub head, but you need to open it. @julijenv can give you information on how to do it.

lornat75, 2016-10-04:
How to build a VGA cable to connect a screen on the PC104 was mentioned here:

https://github.com/robotology/icub-support/issues/183

In short, they point you to this link: https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/Connect_VGA_to_PC104.pdf

lornat75, 2016-10-05:
Closing, feel free to re-open it in case.

closed 2016-10-05 by lornat75