robotology/mesh-support #301: Missing files in iCubNancy01

traversaro, 2016-09-29:
@serena-ivaldi commented on [Thu Sep 29 2016](https://github.com/robotology/robots-configuration/issues/3)

Dear all

we just updated the robot software/firmware on the pc104, and implemented the changes suggested here: https://github.com/robotology/QA/issues/143

We installed the new robot configuration files, but we found issues and we cannot proceed.

1) when we launch robotinterface on the pc104, the robotinterface looks for robotInterface.ini but in our folder we have instead yarprobotinterface.ini.
We renamed the file to proceed temporarily - let us know if you plan to change it also in the robotinterface.

2) even fixing the file, the robotinterface crashed with a fatal error. It is looking for some configuration files that it can't find and it stops. The file generating the fatal error is: torso-ems5-mec.xml that we don't have in our hardware folder. We saw that other robots have it (Genova04, Darmstadt..) and from the logs it seems it is a new file. Could you add it to our robot too?

Thanks in advance

 @paraschos  @misaki43 @dgoepp 

---

@traversaro commented on [Thu Sep 29 2016](https://github.com/robotology/robots-configuration/issues/3#issuecomment-250542290)

You should launch `yarprobotinterface` in place of the deprecated `robotInterface`, see https://github.com/robotology/QA/issues/109 .

For robot-specific support, it is better to use the `icub-support` repository at https://github.com/robotology/icub-support . I am going to move the issue.

traversaro, 2016-09-29:
Apparantly the missing files were added in https://github.com/robotology/icub-main/pull/365 for all the other `ETH` robot, @marcoaccame probably knows if it was just an oversight or something intentional.

serena-ivaldi, 2016-09-29:
Log file

[ypmrprunlog_29_09_2016_07_59_56.txt](https://github.com/robotology/icub-support/files/501371/ypmrprunlog_29_09_2016_07_59_56.txt)

serena-ivaldi, 2016-09-29:
Update: we found that some configuration files had an incorrect number of parameters in the entries. We removed the extra numbers and it works now. 

We created a pull request to update the files in the main repository.

closed 2018-01-16 by julijenv