robotology/mesh-support #302: Encoder problem in the torso, probably very serious

serena-ivaldi, 2016-09-29:
Dear all,
we are experiencing an hardware issue in the torso, that is causing the torso to fall after calibration or not work at all. Here is a copy/paste of the error message from the yarprobotinterface:

`[ERROR] from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1564s 162m 441u: (code 0x02000008, par16 0x0000 par64 0x0000000000000003) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .`
 
It seems a serious problem and already happened multiple times with serious consequences when stopping the yarprobotinterface (the torso does not park but falls like dead). 

@misaki43 @dogoepp @olivierrochel-inria

julijenv, 2016-10-04:
Hi @serena-ivaldi ,
did you try to launch yarprobotinterface faulted and try to move this joint 3 of the torso??? and check if you have values appearing on yarpmotorgui when moving this specific joint?
If the values are inconsistent, (meaning that at some position the continuity of the value ceases or jumps ) you may have one of the wire of this AEA sensor rip off. this sensor come from inside the robot and logically, you should not have any wire unsoldered a this point.
but this sensor  labelled 4S0 has an extension so check as well the connecotr in the middle. the wire gets out from the black wrapping back spinal cord on the robot and gets out from the top of bottom of the back of icub. You should find the junction at this level (you may have to remove the bottom cover to see better the wires). the ending paert of the wires arrive inside the bottom of the robot.
[image]
on EMS board EB5 - P8 which look like that:
[image]
[image]
Hope this help, let me know
@misaki43 @dogoepp @olivierrochel-inria

misaki43, 2016-10-14:
hello,
sorry for this lack of answer, this error didn't appear again. I just verified the encoders' values when we move the torso, and all seems to be ok. Moreover we move from yarpmotorgui the position directly, all seems ok too.
Do you think we can close this issue and re-open it if it re-append?
(Note: if I remember correctly, the error appeared when we put the robot in a seated position, so may be it will re-append the next time we will do it.)

dogoepp, 2017-04-21:
Today we had the torso have a hardware fault in joint torso_roll, as shown in the screen capture:

[image]

The torso was not moving and was in position control, only the arms that we were moving (testing demo force cotnrol).
Just for information, we still have the shaky legs that make the whole robot vibrate a bit (we will open an issue for this matter).

serena-ivaldi, 2017-04-21:
Hello, the error appeared again, while the robot had only started up and was not moving. Here is the error reported by yarprobotinterface:

```
from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 6876s 868m 440u: (code 0x02000008,
par16 0x0000 par64 0x0000000000000008) -> MC: AEA encoder invalid data. Hardware problem
in the magnetic position sensor of the joint caused invalid position readings.
par16 = AEA port (msb) and ID of joint (lsb). + .
```

The problem seems to be an hardware fault on the second joint. Does the encoder need to be replaced ?

@julijenv

julijenv, 2017-04-21:
HI @serena-ivaldi , and @dogoepp ,
the AEA encoder is sending no or invalid data, from my understanding and experience I would suggest you to follow the encoder cable at the elevl of the ebelly and find out if one of the wires is disconnected,, unsoldered or pinch at tsome point on the main electrical wire.
do continuity test in between the AEA pin( pads) and the connector which goes to the bottom poart of icub.
let us know,

serena-ivaldi, 2017-04-21:
We found this in a warning:

```
3237	159,326138	WARNING	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 197s 796m 164u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000014) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
```

and the robot had an impulsive motion, very weird. 

@julijenv where is this cable I need to check? I would rule out disconnected because the torso starts, this error just appears after a while..

serena-ivaldi, 2017-04-21:
This is a plot of the spike - you can see that it suddenly move of 12 degrees and comes back immediately to 0. This causes a problem in the other joints too.

[image]

julijenv, 2017-04-21:
Hi @serena-ivaldi ,
always the same, the patata joint. follow down the wire extremities to find which fo the micrio wires inside it are broken/ almost broken.

serena-ivaldi, 2017-04-21:
The "patata joint"?? 🐙 

Ok I checked the cables but I don't see anything broken... can you show me where I should look better? 

[image]
[image]

serena-ivaldi, 2017-04-24:
Hello, we had again the problem, this time the shaking was smaller so it was a warning but the joints didn't go in idle or hardware fault.

```
2920	3192,165138	WARNING	 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 459s 665m 694u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000008) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
```

@julijenv can you point where we should open the robot to check the cable? As from the pics I really cannot see anything broken.

julijenv, 2017-04-26:
Hi @serena-ivaldi ,
those pictures should help you to find out where to get the end of the patata joint.
The spike joint is the second one - j1 (3 joint in the waist W_3S0 = j0, W_3S1 = j1 and W_4S0 = j2).
Do a continuity test from source to connector.
[image]
[image]
[image]
LEt me know

serena-ivaldi, 2017-05-01:
Hi @julijenv 
where are these located? How can we access them? Should we open the torso in the front or in the back?

julijenv, 2017-05-02:
Hi @serena-ivaldi and @dogoepp ,
here some more details pictures fothe electronics on the robot.
[image]
So after verification your joint 1 disable is related to the sensor 3S0 below the plate of the belly (patata joint ) the connector is connected to the board EB5 (EMS5):
[image]
Hope that will be enough,
I am always availabl epoon skyoe if ever needed.
best of luck

serena-ivaldi, 2017-05-05:
[image]
[image]
[image]

Hi @julijenv 
thanks for the pics! we opened the bottom and found the EB5, here's the picture. The cables seem fine....

serena-ivaldi, 2017-05-05:
To test the 3S0 we have to detach it from the plate, but it seems the card is taped with scotch inside the plate. Is it safe to detach it removing the tape?

julijenv, 2017-05-05:
Hi @serena-ivaldi and @dogoepp ,,
There is NO tape behind the only things are the 2 screws and maybe some soft glue which solidificate and get attached to the plate. I think yu can also try to  touch what ever pin of the chip and find out where are connected every wire to the chip without having to unmount the 3S0 from the plate ( BUT if you do it is better because you can unsure you that the wires are firmly soldered on 3S0 and there isnt any one of them  almost broken).
let me know

serena-ivaldi, 2017-05-05:
Hi @julijenv we tested with the tester every wire (white green yellow red and black) and the connections are all good.

:,(

serena-ivaldi, 2017-05-05:
Hi @julijenv 
the test before was with the 3S0 and everything was ok. We tested the 3S1 (putting the robot at 90 degrees in the torso pitch) and we tested the 3S1. We found that the yellow cable (CS) was not properly soldered. So now @dogoepp is going to solder the yellow cable in 3S1. Hopefully this will solve the problem. 
Thanks for your help! We'll confirm in a few if it works and the problem is solved.

dogoepp, 2017-05-05:
Hello,

The soldering was done. It seems to have solved the problem for now but we have to keep watching this joint for some time.

I'm closing the issue for the time and would re-open it if the problem arises again.

misaki43, 2017-06-12:
Hello,

It seems that the problem appears again. I was moving the arm robot smoothly (that whas in mode "zero torque") and at one moment, the robot's chest fall.

```
960 1519,198651 <ERROR>  from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1594s 481m 606u: (code 0x02000008, par16 0x0000 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
 
2961 1519,551382 <WARNING>  from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1594s 833m 559u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000004) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
 
2965 1524,201753 <ERROR>  from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1599s 481m 613u: (code 0x02000008, par16 0x0000 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
 
 
2969 1528,128433 <WARNING>  from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 1603s 423m 684u: (code 0x0000000b, par16 0x01cf par64 0x00190022000700bb) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
 
2970 1529,198689 <ERROR>  from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1604s 482m 644u: (code 0x02000008, par16 0x0000 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
 
2971 1532,692211 <DEBUG> cfw2can [0] thread ran 595 times, req.dT:10[ms], av.dT:10.08[ms] av.loopT :0.02[ms]
```

At the same time , we had a problem with the board communication that we haven't seen during it utilization (because it was in DEBUG information) but I think it is not linked with this torso problem.
Here is the total log file: 
[log_pc104_yarprobotinterface_3145.txt](https://github.com/robotology/icub-support/files/1068085/log_pc104_yarprobotinterface_3145.txt)

@dogoepp @serena-ivaldi @julijenv @InriaBrice & others

julijenv, 2017-06-12:
Hi @misaki43 , @dogoepp , @serena-ivaldi ,  @InriaBrice & others,
Could you please check the AEA encoder for the joint 0 of the torso - from my understanding it is the YAW joint.
the encoder name is 4S0. The sensor is inside the torso but in the upper part of the chest, the sensor gets out of the ombilical cord form the back of the robot. and have a connector before goint to the board 10.0.0.5.
Please check the integrity of those segment of wires.
[image]

thx in advance

serena-ivaldi, 2017-06-14:
Hi @julijenv can you send a photo highlighting where to look on the robot?

By the way, the error arrived again, and this time we have spikes (like before the soldering fix).

'''
from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 1579s 465m 728u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
'''

See attached yarplog.

[torso_pblm_spike_14_06.txt](https://github.com/robotology/icub-support/files/1074118/torso_pblm_spike_14_06.txt)

julijenv, 2017-06-14:
Hi @serena-ivaldi ,
if you remove the bottom cover or have a view from the back of the robot (see picture) you should be able to see the connector of the sensor 4S0 - labelled- or go upto the EB5 (see pictures above quoted in the issue)
[image]
I'm available on skype anytime, come and look for me

gabrielenava, 2017-06-14:
FYI: this seems the same problem we experienced on the green iCub: https://github.com/robotology/icub-support/issues/473

and we also had spikes in the encoder:

```
[WARNING] from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 530s 724m 859u: (code 0x0200000a, par16 0x0003 par64 0x0000000000000001) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .

```
the problem in our case was a broken cable. Maybe there is a damaged cable that sometimes is causing the problem, but if it is not clearly visible may be difficult to find and fix it.

serena-ivaldi, 2017-06-14:
Hi Julien, here is Dorian,

We started the robot again and moved the torso on the three axes. This did not trigger the error but I don't feel that it is enough to consider this problem solved. The problem has indeed not been found yet.

julijenv, 2017-06-14:
Hi @serena-ivaldi , @misaki43 , @dogoepp ,
please consider making some high range of torso movement to clear this issue and see if the problem pop up again, or in the other way around just leave the robot in some position and move the connectors at the level of EB5 or at the branch to see if the robot go in Hardware fault, this should be convinced you.

dogoepp, 2017-06-16:
High range motions were done two days ago, at startup and after leaving the robot for some time. Nothing observed. The second test still has to be done.

julijenv, 2017-06-16:
Hi @dogoepp ,
it seems that one of the connector was not properly plugged in the best way possible, I'm glad to hear that the behaviour of the robot looks more stable.

dogoepp, 2017-06-16:
Is there no other possibility ? I don't feel that I did much. The only intervention was to unplug and plug again the connector on the EB5 board but I could not see any difference between before and after.

gabrielenava, 2017-06-26:
@julijenv found the broken cable that was causing the encoder failure. We can close this issue

misaki43, 2017-08-31:
Hello,

bad news, the issue appears again!
Yesterday someone has putted the robot on an "half sitting" position (I can c/p the angle position if needed). Then, the robot's torso fall again... The log hasn't been saved.
Today, the problem appears again without doing anything on the robot. Here is the log:

[torsopblm_31_08_2017_14_32_11.txt](https://github.com/robotology/icub-support/files/1267076/torsopblm_31_08_2017_14_32_11.txt)

Could you help us again please? If it is again a broken cable, have you an idea what is happening?
Thanks!

@serena-ivaldi @dogoepp @InriaBrice @julijenv

julijenv, 2017-08-31:
Hi @serena-ivaldi , @misaki43 , @dogoepp ,
to me the real problem is this one:

> <#STRING_START#>cfw2can [0] have not heard from board 1 (channel 0) since 7.79 seconds
<#STRING_END#>
8373
2348,762661
4
<#STRING_START#>cfw2can [0] have not heard from board 1 (channel 1) since 7.79 seconds
<#STRING_END#>
8374
2348,762676
4
<#STRING_START#>cfw2can [0] have not heard from board 2 (channel 0) since 7.82 seconds
<#STRING_END#>
8375
2348,762691
4
<#STRING_START#>cfw2can [0] have not heard from board 2 (channel 1) since 7.82 seconds
<#STRING_END#>
8376
2348,762705
4
<#STRING_START#>cfw2can [0] have not heard from board 3 (channel 0) since 7.81 seconds
<#STRING_END#>
8377
2348,762722
4
<#STRING_START#>cfw2can [0] have not heard from board 3 (channel 1) since 7.81 seconds

This message pop up a lot of time, and if it is true it means that the communication is not passing on can 0 head to torso line, so I guess something got disconnected or floating somewhere.
And before to get the message of encoder you already push the red button, so it is hard to guess what is really wrong.
could you please try to check the connections around every part that could have in contact to something during the half sitting postion.
Please send a video of the calibration phase, it is not clear to me what is really happening.
thx in advance

misaki43, 2017-10-10:
Hi @julijenv,
sorry for the delay. This issue didn't appear only when the robot is in half-sitting position. It seems to be totally random. Often, it happens ~ 1/2 times per day. Here are the last 2 logs.
[torso_pblm_10_10_17.txt](https://github.com/robotology/icub-support/files/1371185/torso_pblm_10_10_17.txt)
[pblm_torso-02_10.txt](https://github.com/robotology/icub-support/files/1371188/pblm_torso-02_10.txt)

These time, the robot was on its its pole, without touching anything. 
Anyway, this time iCubGui was launched and I saw that when the usual issue appears (magnetic position [...]), the robot was shaking and big forces on its foot was visible.

>  could you please try to check the connections around every part that could have in contact to something during the half sitting position
@dogoepp

dogoepp, 2017-10-10:
Hi @julijenv,
Where could I locate this can 0 line ? By the way, is there a document showing which cables go to which boards and with which names ?

I am willing to check all connections around the legs (related to half-sitting) but why would it affect the torso CAN 0 ? In this posture, only the two feet touch anything, namely the floor.

julijenv, 2017-10-10:
Hi Dorian,
the answer will be quite easy, unfortunately when you try to sit iCub on anything it is more than possible that the wires inside the bottom cover got bent (actually they are exposed ) so they could be affected by any touch to the bottom part of iCub.
here ther eare some pdf file of the connection inside your robot:
[iCub 3_3.0.2_Motor Board Placement.pdf](https://github.com/robotology/icub-support/files/1372441/iCub.3_3.0.2_Motor.Board.Placement.pdf)
[iCub 3_3.0.4_5672_Logic.pdf](https://github.com/robotology/icub-support/files/1372442/iCub.3_3.0.4_5672_Logic.pdf)
Please refer to it went looking for any misbehaviour of your robot.

julijenv, 2017-10-10:
From what I could identify on the log file,

> from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 3603s 910m 264u: (code 0x02000008, par16 0x0000 par64 0x0000000000000009) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .

it is more than possible that the AEA sensor connector has been touched during your experiments, you may have to check it.

julijenv, 2018-01-16:
Hi @serena-ivaldi , @misaki43 , @dogoepp ,
any feedback ??!?!?

serena-ivaldi, 2018-01-30:
Hi Julien, sorry for the radio silence. Dorian had checked the cables and found nothing. Now the problem re-appeared today while doing the red-ball demo, and we haven't done any more sitting....

`55367	609,076733	ERROR	 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 812s 482m 758u: (code 0x02000008, par16 0x0000 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .`

julijenv, 2018-01-30:
Hi @serena-ivaldi , @misaki43 , @dogoepp ,
looks like there is a remapping inside the torso joint,
Please make a basic trial, and remove the AEA encoder 3S0 from the board inside the bottom of the robot. you should find it on the left side of the robtot looking from the back of the robot.
If after having removed it, the same equal message pop up , I would conclude that it is the sensor with problems.
let me nkow

julijenv, 2018-09-10:
It would be nice to not just reopened it but fiurst make sure you've tried my last suggestion or let me know more about what is going on, and for which kind of experiment the error raised again?
@serena-ivaldi , @dogoepp , @pi-q , @misaki43 .

pi-q, 2018-09-19:
Yes indeed, sorry for the lack of immediate follow up. 

The problem appeared again, every time we were using the robot, probably about 10 - 60 minutes after starting up the robot (depending on what we were doing; running a torque-controller seemed to make it worse). Letting the robot rest on the pole a few hours would temporarily fix the problem.

We opened the torso and checked the connections, following the instructions above. We think we found the problem and how to fix it. It's a problem with the connectors between the AEA sensors and the EB5. When the robot performs some torso movement after the motors are on for some time (might be due the heat), the wires get bent in a way that causes an interruption of the connection. We just have to manually reposition the EB5 and push/plug properly the connectors as you said.

closed 2018-09-19 by pi-q