robotology/mesh-support #304: Need a guide to create all folders and files for new real robots

misaki43, 2016-10-03:
Hello @DanielePucci @traversaro and others,

We have a lot of problems when we try to launch our robot icubNancy01, because some folders and files that are specific to our robot don't exist.
We began to launch a lot of issues because of that, but I think it would be more efficient to do a kind of guide that explain what files and folders we need to create, where we have to place them and how we can fill them.
I know you will loose time with that, but I think at the end you will win times too (the others laboratories that will install their robot will be able to do it in a more autonomous way).

We can do it together on Skype (I launch programs, tell you all things that are missing, you tell me what to do and we create at the same time the tutorial)

is it possible?
@serena-ivaldi @olivierrochel-inria @dogoepp

traversaro, 2016-10-03:
Hi @misaki43 , 
   I completely agree with you, but I think that this is just a problem of `CoDyCo` related software, right? 
   All the other software that you can get with `icub-main` should have all their configuration files available in https://github.com/robotology/robots-configuration , and IIT is supposed to update them. If this was not the case (see https://github.com/robotology/icub-support/issues/301) I am sorry, but that was an oversight, not a systematic problem. 

> We began to launch a lot of issues because of that but I think it would be more efficient to do a kind of guide that explain what files and folders we need to create, where we have to place them and how we can fill them.

Actually, at the moment in `CoDyCo`-software repository there are no issue open for missing files for `iCubNancy01`. You sent some email in the past week, but please open issues for this kind of problems. 

> We can do it together on Skype (I launch programs, tell you all things that are missing, you tell me what to do and we create at the same time the tutorial)

At least for me, it is more convenient to do this offline: you take note of the missing files, and then we add them/write a document on how to add them. But again, this is a `CoDyCo`-software related issue, so if for you it is ok I will open an issue in the `codyco-superbuild` repo.

closed 2016-10-03 by traversaro

traversaro, 2016-10-03:
See https://github.com/robotology/codyco-superbuild/issues/155 .