robotology/mesh-support #306: Broken pole lock, replacement request

hu-yue, 2016-10-11:
### Description of the failure

The cable of the pole lock is broken.

### Detailed conditions and logs of the failure
As in the picture below, the cable connecting the lock (see red circles) of the pole is broken. 

[image]

Since we are not able to repair this kind of cable, we would like to ask if it is possible to have a replacement of the piece of cable? 
Thanks!

julijenv, 2016-10-13:
hi @hu-yue ,
we will build a new one and send it to you.
best regards , Iwill let you know, please close the issue when you will remounted it.

hu-yue, 2016-10-13:
@julijenv Sure, I wait for your update then! Thank you very much!

spada1978, 2016-10-14:
hi
 I realized the pole cable.
if I communicate the shipping address, you will be sent as soon as possible

hu-yue, 2016-10-14:
@spada1978 Thank you very much! 
The shipping address is as follows:
Yue Hu, 
ORB, Heidelberg University 
Berliner Str 43, 
69120, Heidelberg
Germany

hu-yue, 2016-10-20:
@spada1978 @julijenv the cable has arrived and we have already mounted the lock back to the pole. Closing the issue.Thanks again!

closed 2016-10-20 by hu-yue

spada1978, 2016-10-20:
Perfect...
see you