robotology/mesh-support #307: No prosuppination control when arm joints are in torque mode

dcam0050, 2016-10-12:
### Description of the failure
No control of prosuppination when putting the joints in torque mode

### Detailed conditions and logs of the failure
I launch `ARE` with `impedance = on` and the iCub fails to control the prosuppination of both arms.

julijenv, 2016-10-13:
Hi @dcam0050 
More details would be more appreciated, thx in advce

dcam0050, 2016-10-13:
I was trying to locate what the problem is within the xml files in iCubSheffield01/hardware/motorControl and found that the stiffness and damping values of left lower arm and right lower arm are 0 for all joints so I'm thinking this is the problem. I'm not sure what value to set them to though.

I've recorded a video of what the iCub is doing here : https://www.youtube.com/watch?v=Byk9nus0yJk (unlisted video) to better convey what is happening. In the video I first issue a `home all` command to `ARE` rpc port followed by an `expect` command and then a `give` command.

While it is doing the home command, the prosupination angle is seen to drift a bit but slowly. The problem is more prominent with the `expect` and the `give` commands where the palm should be face up

pattacini, 2016-10-13:
Hi @dcam0050,

Zeros for `stiffness` and `damping` in the xml are just fine, because these parameters are meant to be tuned at user level (checked with @randaz81 right now).

Therefore, it's `ARE` in charge of setting those values: see `impedence_arm_stiffness` and `impedence_arm_damping` options within the `config.ini` file.

pattacini, 2016-10-13:
You could verify the values set by `ARE` using `yarpmotorgui`.

dcam0050, 2016-10-13:
The settings in `ARE config.ini` are as follows:
```
impedence_arm_stiffness             (0.4      0.4       0.4     0.2       0.05)
impedence_arm_damping            (0.03    0.03    0.03    0.01    0.0)
```

I've verified that the values for the first 4 columns are set correctly via `yarpmotorgui` but the last column which is prosupination, when checking via yarpmotorgui the settings are `stiffnes = 0` and `damping = 0`

I also tried setting the values manually, then pressing `send` and then pressing `refresh` to see if the values stick or not but I cannot set them manually either

pattacini, 2016-10-13:
Ok then, this must be the problem... thanks @dcam0050 for checking.

@randaz81 @julijenv any idea?

dcam0050, 2016-10-13:
Last time we had this in Sheffield a firmware update was suggested but the firmware now is very recent (2weeks old) and all libraries are up to date

dcam0050, 2016-10-13:
When sending a small value for the stiffness (<0.1) I don't need to press `refresh` in order to see that the setting was not applies even though the `min` field specifies 0.002 as the minimum

But if I send a stiffness > 0.1 the joint moves and `current = 0.071` but another `refresh` sends it to 0

The value for `damping` on the other hand I can modify and send without a problem

dcam0050, 2016-10-17:
Any news regarding this issue please?

valegagge, 2016-10-18:
I @dcam0050,

I'll check your problem on our robot in lab asap.

pattacini, 2016-10-25:
@valegagge did you manage to reproduce the problem?

pattacini, 2016-11-03:
Let me try to quickly summarize hereinafter the current status, _up to my knowledge_:

- @valegagge committed a fix within the `devel` branches of `icub-main` as well as firmware repositories.
- cherry picking this fix into `master`'s doesn't look straightforward and requires some effort.
- @robotology/firmware-developers are now taking up other high priorities tasks, thus there's very little time we can devote to this.

@dcam0050 given this status, you might have to wait long to get `master` work in this respect. Do you want to give a try to `devel` then?

randaz81, 2016-11-03:
Beware `devel` branch is unstable, it may cause other malfunctioning, if you decide to go in this direction, use it your own risk. No support is given on devel branch issues.

pattacini, 2017-02-18:
Closing this, @dcam0050 feel free to report more on this.

closed 2017-02-18 by pattacini