robotology/mesh-support #308: Error during the calibration of the Left_Arm

misaki43, 2016-10-12:
Hello,

every time during the calibration of the robot, there are errors linked to the left arm of our robot. After the calibration, its left fingers are idled, and we cannot control them on yarpmotorgui.
Could you help me to solve this problem? 
(Tuesday we have some demo to launch, it would be great if we could use the arm)
Note: The left arm has been take off and reassemble.

Here are the logs of the errors

> 7653		3838.372401		ERROR		Left_Arm_Calibrator : joint  3  is idle, skipping park 
> 7652		3838.372371		ERROR		Left_Arm_Calibrator : joint  2  is idle, skipping park 
> 7651		3838.372338		ERROR		Left_Arm_Calibrator : joint  1  is idle, skipping park 
> 7650		3838.372305		ERROR		Left_Arm_Calibrator : joint  0  is idle, skipping park 
> 968		41.954070		ERROR		Left_Arm_Calibrator : set 1 : some axis got timeout while reaching zero position... disabling this set of axes
> 965		41.954022		ERROR		Left_Arm_Calibrator : checkGoneToZeroThreshold: joint  2  Timeout while going to zero! 
>

julijenv, 2016-10-13:
Hi @misaki43 ,7
the messages log lines you sent to me are the last messages on the yarprobotInterface log but doesnt show anything particular (as a matter of fact you are asking for closing the interface so if something append you may unconoutered this kind of messae even coming from some other joint).
So now, please send me the entire logfile (launch on pc104 shell : yarprobotInterface &> log_entire.txt).
Launch yarprobomotorgui, when the robot looks calibrated please send me a picture of the window of the left arm (to see whoich joints went idle).
join me both the picture and logfile.
and i will suggest you what to do,

misaki43, 2016-10-13:
Hello,
ok thanks for your help!
[Here](https://drive.google.com/file/d/0B9sXstBzNOiuTGsyYmU5REtBR28/view?usp=sharing) is the screenshot and here are the logs (but I launched the robot a second time with yarplogger to be more readable if you want, [here.](https://drive.google.com/file/d/0B9sXstBzNOiuXzJzMzdOQlByNjA/view)):
```
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/icub/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/yarprobotinterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./icub_all.xml
yarp: Port /log/pc104/yarprobotinterface/3133 active at tcp://10.0.0.2:10008
yarp: Sending output from /log/pc104/yarprobotinterface/3133 to /yarplogger using tcp
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./icub_all.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of iCubNancy01 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . Import a file for a different robot build. Found 0 instead of 1 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of iCubNancy01 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . Import a file for a different robot build. Found 0 instead of 1 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/head-cfw2_can0-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./wrappers/motorControl/head-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/torso-ems5-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/torso-ems5-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/pc104.xml 
[... cut ...]
[[01;32mDEBUG[00m]Left_Arm_Calibrator : Moving to park position, joint: 4 
[[01;32mDEBUG[00m]Left_Arm_Calibrator : Moving to park position, joint: 5 
[[01;32mDEBUG[00m]Left_Arm_Calibrator : Moving to park position, joint: 6 
[[01;33mWARNING[00m] from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 401s 426m 468u: (code 0x0000000d, par16 0x002b par64 0x000c009701540006) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of T

pattacini, 2016-10-13:
Hi @misaki43,

Next time, could you please attach a file containing the log instead of copying + pasting the log itself herein? We would gain a lot in readability 😄

misaki43, 2016-10-13:
Hi @pattacini, it is better like that, thanks!

julijenv, 2016-10-13:
Hi @misaki43 ,
I dont understand why your FINGERS dont work, are you sure it isnt the 4 first joint of the left arm that are not working???
secondly, could you try to fit more appropriately the windows in order to see better the values of the encoders for the 4 first joints? from the log joint 2 is read as 143 which is higher than the range allowed so i guess that this encoder has an issue.
my suggestion would be to relaunched the robot try to fit better windows in order to see the proper values of the encoder and try to move the arm with your own hands and check if the values change.
let me know

misaki43, 2016-10-13:
Oh sorry @julijenv , I stupidly wrote what we said to me without having verified it: the problem is not the fingers, but the 4 first joints of the left arm, as you noticed.
Here is a better screen-shot that shows the encoders.
Before you recommended it, I avoided to move the arm for fear of forcing it to move. Now I just tried to move it from yarpmotorgui but I cannot move it (see joint pictures).
[image]
[image]

julijenv, 2016-10-13:
Hi @misaki43 ,
please to get in touch with me on skype: my ID is icub.support
in the meanwhile please make sure that you install teamviewer. and make sure you will have skype on a laptop on which you will be able to plug a webcam to let me know what is going on.
thx,

misaki43, 2016-10-17:
Hi Julien,
we have updated the robot-configuration files for our robot. We fixed a typo in one xml file (pull request coming soon). We were able to launch the yarprobotinterface with the fault button pressed, with the two parameters skipcalibration=true and useRawEncoderValues=true.
We checked if the encoders' values change correctly when we move the joints of the robot one by one, and it seems to be working fine. We didn't spot any incongruencies.

However, we then launched the yarprobotinterface enabling calibration and rawEncoder=false (i.e., the normal setting for launching the robot) and unfortuntaley we had to stop the robot because of a calibration issue in the right arm, that could harm the robot's covers:
precisely, the right arm joint n.2 (shoulder yaw) is turning outwards (i.e., the arms turns in the other direction of the robot's body) in such a way that the back cover of the upper arm touches the cover of the torso and pushes. We were afraid that this could break eventually the covers, so stopped the robot.

It seems to be an error or a bug coming from some wrong configuration parameter. We opened the configuration files of the calibration (xml files) and compared to other robots such as Genova02 and Darmstadt, but we couldn't find major changes that points us towards the issue. 

In attachment, the log of the yarprobotinterface from the yarplogger.

Amy idea of how to fix this issue in the calibration?

Thanks in advance

[ypmrprunlog_17_10_2016_rightarm_calibration_issue_joint2.txt](https://github.com/robotology/icub-support/files/534112/ypmrprunlog_17_10_2016_rightarm_calibration_issue_joint2.txt)

misaki43, 2016-10-25:
Here are the logs with our left arm issue
[log_entire.txt](https://github.com/robotology/icub-support/files/550652/log_entire.txt)

misaki43, 2016-10-25:
[logfile01.txt](https://github.com/robotology/icub-support/files/550724/logfile01.txt)

misaki43, 2016-10-25:
[iCubNancy01.zip](https://github.com/robotology/icub-support/files/550740/iCubNancy01.zip)

misaki43, 2016-10-25:
shoulder position "stand to position"
[icub-shoulder.zip](https://github.com/robotology/icub-support/files/550968/icub-shoulder.zip)

marcoaccame, 2016-11-08:
After some work of @julijenv, @ale-git and myself, I think we have solved the problem.  We tested the solution today on left arm and after the cable in right arm is fixed (see issue #325) we shall test the whole robot and commit changes.

See next post for details.

marcoaccame, 2016-11-08:
After some skype session we saw that the joint 2 in both arms is not at correct position after calibration. 

A movement command of x degrees ordered by the yarpmotorgui effectively moves of x degrees but ... there is a strange constant offset. 

After much investigation, we solved in this way: 

1. we corrected a bug in the calibration xls file: the sign of the zero of joint 2 had a wrong sign for both arms.
2. we acquired values again from yarpmotorgui (see correct procedure later on), applied them on the xls file 
3. corrected the values of [calibration3, calibrationZero, calibrationDelta] in the xml calibration files of both arms.
4. verified correct positioning of left arm. Right arm has a broken cable and will be verified after its replacement.

Here are more details.

Analysis of the calib xml files for left and right shoulders have shown strange values. 
As a first thing, the parameter "calibrationDelta" is too large. It normally should be 0 or contain a small correction of a few degrees. Then, comparison with the excel files shows following differences:
- calibrationZero: values in xml and in excel are different. They must be equal as they are independent of the encoder readings and depend only on mechanics.
- calibration3: values in xml and in excel are different. They are allowed to be different because they depend on the readings of the encoders after the robot's arms are positioned aligned to the body. However, the values appear to be quite different.
  
Following the above considerations, we decided to perform new readings of the encoders and to recompute the calibration values from the excel file.

However, this operation did not solve the problem. We investigated on possible bugs inside the FW, but we did not find anything strange. Then, we thought that the excel might be been wrong, thus we verified its formulas vs the FW implementation. But we found nothing strange.

Finally, we found out.

In the excel file there was a wrong sign in the fixed values of the zeros. In V1 mechaniscs, joint 2 on left arm must have value -180 (and not +180) and joint 3 on rigth arm must have value +180 (and not -180). The values on the excel and on the xml files were the ones of V2 mechanics.

At this point we started again: we applied the readings of the encoders to the excel file, wrote results into the relevant xml calibration files and we successfully removed the constant offset in the left arm.

After #222 is solved (broken cable on right arm), we shall test both arms, commit new excel and xml files and hopefully close the issue.

Here is how to get the values of encoders to be put inside the excel and then the xml files.

1. edit general.xml and set skipCalibration and useRawEncoderData to true.
2. make sure that the ETH board reads the encoder values and send them up without any transformation by doing: 
   - impose a singular matrix "matrixE2J" inside files hardware/motorControl/left_upper_arm-ems1-mc-service.xml
     and hardware/motorControl/right_upper_arm-ems3-mc-service.xml.
3. power off and then on the motors (so that any previous command from yarprobotinterface is reset).
4. start yarprobotinterface with fault button pressed.
5. read values in yarpmotorgui and put them inside the excel file.
6. stop the robot.
7. edit the files calibrators/left_arm-calib.xml and calibrators/right_arm-calib.xml by filling variables calibration3 and 
   calibrationZero and keeping calibrationDelta = 0.
8. revert files hardware/motorControl/left_upper_arm-ems1-mc-service.xml and hardware/motorControl/right_upper_arm-ems3-mc-service.xml
   to have the correct matrixE2J matrix. Revert general.xml file.
9. test effectiveness of the calibration with the fault button pressed.

marcoaccame, 2016-11-09:
After closure of issue #325, today I have completed the calibration of both arm.

Now everything is committed in robots-configurations and @misaki43 and I have verified that the xml configurations in repository works correctly.

What remains to be done is the fine calibration, task which I assign back to @julijenv.

misaki43, 2016-11-10:
The fine calibration has been done with @julijenv. The issue is solved, thanks a lot!
 @marcoaccame @olivierrochel-inria @dogoepp @serena-ivaldi

closed 2016-11-10 by misaki43