robotology/mesh-support #310: Accelerometers and gyroscope not working in the left hand/forearm

massimoregoli, 2016-10-13:
### Description of the failure

The data sent through the '/icub/left_hand/intertialMTB'  port by the accelerometers/gyroscope of the left hand/forearm of the purple iCub never change over time (there is not even noise).

### Detailed conditions and logs of the failure

The problem occurs ALMOST every time that `yarprobotinterface` is run. Only once we noticed that the accelerometer and the gyroscope of the hand (but NOT the accelerometers of the forearm) were apparently back to work, but the problem occurred again after the next run of `yarprobotinterface`.

marcoaccame, 2016-10-17:
I have talked to @massimoregoli when he came to me for help and I asked him some more details.

Here they are.

* The reading on the YARP port is present with a valid timestamp, but then the values and the timestamp of a particular sensor are not updated anymore (but the other change).
* Moreover, the MTB boards which do not give a reading anymore don't blink as the others. Do they not blink at all or do they blink differently?

I suspect that the problem is due to one (or more) MTB board(s) which stop working.

nunoguedelha, 2016-10-19:
@marcoaccame, the faulty boards don't blink at all. They steady blue and red less light up.

nunoguedelha, 2016-10-19:
@francesco-romano F.Y.I . As per @marcoaccame the problem is really with the boards (he's doing some last checks, also on the firmware). In that case we will ask @julijenv to take a look.

julijenv, 2016-10-19:
Hi @nunoguedelha , one simple question just pop up in my mind:
Do the sensors from the right arm work properly?
if yes, try to check if the formware is the same on the left arm and the right arm.
if they are identical, I presume that i might have a hardware problem either at the level of the boards or an electrical wire.

nunoguedelha, 2016-10-19:
@julijenv , the sensors on the other limbs (right arm, left and right leg) work properly. The protocol & firmware are the same for all the parts 'PROTOCOL: 1.0    -     FIRMWARE: 2.17.1'.
@marcoaccame please correct me if I'm wrong.

julijenv, 2016-10-19:
Hi @nunoguedelha , ill check the connection tomorros, and let u know ;)

marcoaccame, 2016-10-21:
Hi @nunoguedelha ,

it is correct: the latest MTB's FW has protocol 1.0 and firmware 2.17.1

Let me write also what I have suggested to you for checking the presence of MTBs at start of yarprobotinterface:
* In files hardware/skin/xxxxx-intertial_mbt.xml, section SERVICE.PROPERTIES.CANBOARDS set the value of PROTOCOL to 1.0 (and FIRMWARE 2.17.1).
* With such a configuration, the EMS board checks the presence of the MTB boards when yarprobotinteface starts the device.
* If any board is not present or not working or with a mismatching protocol / firmware version, then there is a diagnostics message and a failure to launch the device.
* In case of missing MTB boards, you can keep on working with the others just removing some from section SETTINGS.enabledSensors

You also raised a good point when you asked: 'What if the MTB crashes after start of yarprobotinterface? Can't we check its presence also in runtime?".

Well, following you suggestion I have implemented this check. See https://github.com/robotology/icub-firmware/commit/ad820d804fda213d910fd5c3327d089adef0826f

I haven't produced yet an official FW in icub-firmware-build (I did one just yesterday) but in case of strange behaviours I can give you a test FW with this feature on.

nunoguedelha, 2016-10-21:
Hi @marcoaccame , thanks a lot for all the information. I'll let you know if I get other issues and need the monitoring.

closed 2018-04-09 by julijenv