robotology/mesh-support #311: Joint 2 (Neck Yaw) goes in hardware fault when performing red ball demo

S-Dafarra, 2016-10-13:
### Description of the failure
When performing the redBallDemo on the purple robot, the neck yaw joint goes in hardware fault. It does not happen immediately, but almost all the time. By setting idle->run in the motorgui it is possible to move it again. 
With the motorgui we tried to move it at different speeds, but the issue does not occur. 
During the demo, the joint should be in "velocity" mode.

pattacini, 2016-10-13:
Out of sheer curiosity, how did you perform the red-ball demo with Purpley, given that it's not equipped with standard cameras? 😃 
Was it a _peculiar_ red-ball demo?

By the way, we recently found out - at least on Blacky - that the minimum bound of the pitch joint was a bit too low causing indeed frequent HW faults. We nailed it down as per https://github.com/robotology/robots-configuration/commit/c2bf98c542b74b7e7b73e2ed6c4e26007dc4f5a9.

As final remark, the gaze controller should drive the joints in `position-direct` mode instead of `velocity`, according to latest control standard. Therefore, I'd suggest you to check this out.

pattacini, 2016-10-13:
Just realized your joint was the yaw, while ours was the pitch, hence the problem must be different.

iron76, 2016-10-14:
@randaz81 thinks that the problem is related to an "over current" fault. He checked the issue some time ago but the current he measured were not abnormal and were not exceeding the current thresholds. He will implement a *temporary* solution by disabling this over-current fault.

arrenglover, 2017-06-06:
I know this issue is old, but it still hasn't been closed and I'm getting the same problem. Using the gazeController to look at a stereo point, the yaw continually goes into hardware fault (within 10s of operation). I'm setting the eye and neck trajectory time to be larger (i.e. slower motions). Is there an update on either the robots configuration, or the over-current fault on this issue?

closed 2019-10-01 by julijenv