robotology/mesh-support #314: joint nr. 10 (thumb flex) on left arm not working

frank-foerster, 2016-10-17:
### Description of the failure
Joint 10 of left arm (thumb 'flexing' joint) does not calibrate and doesn't work once calibration has finished.

### Detailed conditions and logs of the failure
Side note: When testing the new motors and wrist joints in the context of issue  #232 we had to change the signs of `kp`, `kd`, and `ki` in section `POS_PIDS` to negative (file: icub_left_hand.xml). We also changed the sign of the 3rd entry of the `calibration3` line in file `.../calibrators/left_hand_calib.xml` from 6.600 to -6.600. 
As the very same joint is affected now, I'd be surprised it these changes were not relevant. 

Joint 10 most times does not calibrate and can't be actuated via the yarpmotor gui. 
On one single run of yarprobotinterface it flexed but did not move back to the extended (= zero) position and could neither be moved afterwards via yarpmotorgui.
Canloader does not show any problems on Net 1 and Net 5 (all boards are displayed as RUNNING) and the type and number of boards are identical to the equivalent boards on the nets controlling the right hand / arm (nets 2 and 6 respectively).
Also the command line output of the yarpmotorinterface does not show any errors. 
The only suspicious looking log entry I can spot is generated during shutdown of the interface: 

`[ERROR]sharedcan [1] board 3 FAULT OVERLOAD AXIS 0
...
[ERROR]sharedcan [1] board 3 FAULT OVERLOAD AXIS 1
[ERROR]sharedcan [1] board 4 FAULT OVERLOAD AXIS 0
...
[ERROR]sharedcan [1] board 4 FAULT OVERLOAD AXIS 1`

Im not sure though if these messages have anything to do with the given error as the number in brackets after 'sharedcan' as well as the board numbers sometimes differ between different runs of yarprobotinterface. Here's another one (from a different run of the interface).

`[ERROR]sharedcan [6] board 7 FAULT OVERLOAD AXIS 0
...
[DEBUG]sharedcan [6] msg from board 7: PWM DIS:1 
[ERROR]sharedcan [6] board 7 FAULT OVERLOAD AXIS 1`

And another one: 

`[ERROR]sharedcan [6] board 5 FAULT OVERLOAD AXIS 0`

I tried changing the sign back in the above mentioned files but that did not have any effect on the functionality of the joint. 

Any ideas what could cause this and how this could be fixed are highly welcome. 

Full log of one of the runs is here: [yarprobotinterface__log](https://www.dropbox.com/s/9f827lglr28v4hu/yarprobotinterface141016_1.txt?dl=0)

julijenv, 2016-10-17:
Hi @frank-foerster ,
R U sure that the change you've made are correctly affected the files that are really used inside yarprobotInterface? just checked that, becuase on friday everything was working correctly.
So I wonder why now it is not anymore the case.
let me knoiw, or come on skype

frank-foerster, 2016-10-17:
Hi @julijenv ,
it says in the log file that the correct files are being read, that is
`[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./hardware/motorControl/icub_left_hand.xml `
and
`[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./calibrators/left_hand_calib.xml`
and those are the ones we changed. I just double-checked and they do contain the values with the changed sign. 
I do get the following warning under these messages, but these warnings appear to be displayed with every single file that is being read:
`[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./hardware/motorControl/icub_left_hand.xml at line 2 . Unknown document type. Unknown url http://www.icub.org/DTD/robotInterfaceV1.0.dtd 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./hardware/motorControl/icub_left_hand.xml at line 2 . Unknown document type. Unknown url http://www.icub.org/DTD/robotInterfaceV1.0.dtd 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./hardware/motorControl/icub_left_hand.xml at line 2 . Unknown document type. Missing version in Url http://www.icub.org/DTD/robotInterfaceV1.0.dtd 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./hardware/motorControl/icub_left_hand.xml at line 2 . Unknown document type. Missing version in Url http://www.icub.org/DTD/robotInterfaceV1.0.dtd 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./hardware/motorControl/icub_left_hand.xml at line 1 . No DTD found. Assuming version yarprobotinterfaceV1.0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubHertfordshire01/./hardware/motorControl/icub_left_hand.xml at line 3 . Trying to import a file for the wrong robot. Found iCubHertfordshire01 instead of icub `

closed 2018-04-09 by julijenv