robotology/mesh-support #316: iCub Elektro-automatik PSU problem

AlexAntn, 2016-10-17:
Hello,

We have recently had issues with our old PSUs (see  #293 ), and had to replace them with the new Elektro-automatik PSUs. This is on iCub Plymouth01.

We connected them according to the wiring settings provided on [http://wiki.icub.org/wiki/Electrical_wiring](http://wiki.icub.org/wiki/Electrical_wiring), and we tested them successfully when the iCub is not connected.

However, when we connected the iCub, upon switching on the green motor switch (after we had already turned on the CPU switch), the power supplies turn off any output (setting the values to 0).

Some pictures of the connections done:

[image]

[image]

Detail of the grounding connection:
[image]

Any idea what might be the problem, or any tests we can run?

Thanks,
Alex

julijenv, 2016-10-17:
Hi @AlexAntn ,
COuld you please give me the OVP and general settings of every single power supply?
Meanless to say that you have set the power supply for the motors to REMOTE in order to be able to allow the output to synchronize with the switch.
let me know,

AlexAntn, 2016-10-19:
Hello @julijenv ,

The 12V supply is set to:
 
Under voltage 11.5V
Overvoltage 14.5V
 
Voltage 13.5V
Current 20A
Power 500W
 
The 24V supply is set to:
 
Under Voltage 24V
Overvoltage 48V
 
Voltage 24V
 
Current Not set
Power Not set
 
We have taken the switched 12V signal from the Servo Power Switch  to a 12V relay. (on old supply connected to shutdown I/P)  
 
This switch on the relay has normally closed contacts they are connected to the REM-SB (pin 13) and DGND (pin 4) on the 15way D-type
 
So when there is NO 12V signal from the Servo Power Switch (off) the 24V Output is disabled (output on/off Button on front panel disabled)
 
When the Servo Power Switch is on 12V to relay is on switch contacts on relay open Therefore REM-SB floats high.
 
So Output ON/Off Button (front panel PSU) is not disabled and the 24V supply can be switched on via the button.
 
The issue we have is that the Output of the 24V supply is not switched on as soon as the Servo Power switch is switched on it just enables the Front Panel On/Off Button.
 
We have not tried to synchronise the Servo Power Switch with the Front Panel ON/Off button as yet.
 
We turn on the 12V CPU supply everything OK Current draw from 12V supply 3.57A
 
When Servo Power switch turned on 12V switches itself off Output LED on front panel comes on
 
Have tested the 12V Supply on an Electronic Load tester up to The 20A limit set works fine when 20A hit the voltage output falls away (supply does not switch itself off)
 
So the question is does the 12V servo supply and the 24V sevro supply have to come on together at the same time, because our current setup does not do that.

julijenv, 2016-10-24:
Hi @AlexAntn ,
Sorry for being so long in the reply, it was hard to understand the reason of this misfunction but after reading that you changed both Power Supply and see that you even created from scratch the switch in the back of the 24-48V power Supply, some questions pop up to my mind in oreder to understand where is the problem.
First of all, I see ethat you set the 24-48V power supply to 24V which is the rate as the undervoltage setting, you may need to consider to pass the setting to 40V (as we do here).
Secondly, did you create the little board from what design?
here is the board we use (seethe picture below), did  you use the same component? did you wire the same way?
[image]
Let me know

AlexAntn, 2016-11-02:
Hello @julijenv 

Sorry for the delay in answering, time has been scarce recently.... 

Anyway, we did not have that connection (we only had the relay 4016 RL2). We updated this board with that schematic, and now everything works fine with the PSU when it is disconnected from the iCub.

However, when we connect the power to the iCub, the moment we switch the motors on, the two PSU immediatelly turn the output off. 

We suspect that something might have been damaged when the previous PSU had its fuse blown. Any ideas on what we can check to debug this issue?

The full start-up sequence (with the iCub plugged in):

1) We switch both power supply units on;

2) we switch the CPU green switch on, the 12V PSU starts correctly, 40V PSU is still off. iCub has leds on, we can hear the head ventilation running, nothing wrong so far.

3) we switch the Motor green switch on, the 40V PSU starts and immediatelly switches output to off. 12V PSU also switches output to off. iCub switched off all leds, no more ventilation, all power is off.

Thanks in advance!
Alex

julijenv, 2016-11-03:
Could you please try to get in touch with skypeID :icub.support.
thx in advance

AlexAntn, 2016-11-09:
After a skype meeting, we have adjusted the values for UVL, OVP, and power of the power supply units, which made it work.

As a reference, the power was 0.770 KW for the pc104 PSU and 1.5 KW for the servos (lower one). 

the 12V PSU was between 11V and 14.6V
the 24/48V was set at 36V/48V

I will validate these values again the next time we turn the iCub on.

now everything is working, thanks for the support!

closed 2016-11-09 by AlexAntn