robotology/mesh-support #317: Communication to boards issues

paraschos, 2016-10-20:
### Description of the failure

I am getting random communication errors from leg boards. 

### Detailed conditions and logs of the failure

Cold boot: 
[ypmrprunlog_20_10_2016_05_13_43_strange.log.txt](https://github.com/robotology/icub-support/files/542483/ypmrprunlog_20_10_2016_05_13_43_strange.log.txt)

Emergency button *not* pressed:
[ypmrprunlog_20_10_2016_05_18_44_button_not_pressed.log.txt](https://github.com/robotology/icub-support/files/542488/ypmrprunlog_20_10_2016_05_18_44_button_not_pressed.log.txt)

Failure while the robot was standing on the ground (it started to fall and took me a while to press the e-stop as I was not close to it):
[ypmrprunlog_20_10_2016_06_44_5_random_falling.log.txt](https://github.com/robotology/icub-support/files/542504/ypmrprunlog_20_10_2016_06_44_5_random_falling.log.txt)

paraschos, 2016-10-20:
Happened once more. I am testing the F/T for saturation values. First 2 errors (3035/3040) when I placed the robot on the floor for the first time. I lifted and placed the robot down to the floor a couple of times but I didn't get any more errors. The rest of the saturation occurred when I was pushing
the robot a bit, mainly in the saggital plane. Communication issues towards the end of the log.

[ypmrprunlog_20_10_2016_08_28_09_pushing_test.log.txt](https://github.com/robotology/icub-support/files/542607/ypmrprunlog_20_10_2016_08_28_09_pushing_test.log.txt)

marcoaccame, 2016-10-21:
Hi @paraschos,

the first  log `ypmrprunlog_20_10_2016_05_13_43_strange.log.txt`  tells that board `EB2` has detected the outgoing ETH link going down.
That has caused a lot of problems because configuration messages were lost.

Amongst the lost messages there almost surely was one from board `EB3` telling that the incoming ETH link was detected down.

Then, after about two seconds there are messages from `EB2` and `EB3` telling that the links goes up, but the damage was already done.

Most probably, the problem is the cabling between `EB2` and `EB3` which maybe due to a movement gets loose.

Allow me now some time to read the other two log files.

```
from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 63s 636m 261u: 
(code 0x06000001, par16 0x0001 par64 0x0000000000000000) -> 
ETH monitor: link goes down in port specified by par16 (0 = P2, 1 = P3, 2 = internal) + .
```

```
... a  lot of errors due to loss of communication ...
```

```
from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 65s 537m 307u: 
(code 0x06000000, par16 0x0001 par64 0x0000000000000000) ->
 ETH monitor: link goes up in port specified by par16 (0 = P2, 1 = P3, 2 = internal) + .
```

```
from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 65s 536m 521u: 
(code 0x06000000, par16 0x0000 par64 0x0000000000000000) -> 
ETH monitor: link goes up in port specified by par16 (0 = P2, 1 = P3, 2 = internal) + .
```

```
...
```

```
yarpdev: ***ERROR*** driver <embObjMotionControl> was found but could not open
```

paraschos, 2016-10-21:
Thx @marcoaccame for the info. That would mean that `10.0.1.3 - 10.0.1.4 - 10.0.1.5 - 10.0.1.6 - 10.0.1.10 - 10.0.1.7 - 10.0.1.8 - 10.0.1.11 - 10.0.1.9` didn't get the proper configuration messages when the outbound of EB2 went down... I would like to take a look on the cabling. Where can I find the physical locations of boards (if I don't want to start flashing them one by one :) 

One more Q: so does `par16 0x0001` indicate the outbound and the `par16 0x0000` the inbound? Shall we update the error message to be a bit more descriptive?

marcoaccame, 2016-10-21:
Hi @paraschos , here are some answers.

1. yes: if link between EB2 and EB3 is down all the boards you wrote are unreachable.
2. for info about physical location of boards please refer to @robotology/icub-support-team. 
3. about the text printed by the diagnostics message ... P2 and P3 are the connectors on the EMS board and the guys in the support team who are the main receptors of these messages know which these connector are.  
4. about making diagnostics more descriptive: yes, it is something we trying to do little by little.

paraschos, 2016-10-21:
Thx again @marcoaccame!  If you want I can also contribute in the documentation/diagnostics. Send me a pm to chat about it if there is something reasonably small for me to contribute (focusing on issues I have).

@marcoaccame Did you also skim over the rest of the logs?

@robotology/icub-support-team any info for the locations of the boards? Especially how can I inspect the  connecting EB2 P3 -> EB3 P2?

marcoaccame, 2016-10-21:
I have looked at the other logs.

* The `ypmrprunlog_20_10_2016_06_44_5_random_falling.log.txt` shows the same behaviour of the previous log: that the outgoing link of `EB2` goes down and then up several times. 

* The `ypmrprunlog_20_10_2016_05_18_44_button_not_pressed.log.txt` seems OK. You started yarprobotinterface and then you pressed CTRL-C, did you not?

* The `ypmrprunlog_20_10_2016_08_28_09_pushing_test.log.txt` shows some warnings of STRAIN saturation and then ... again: outgoing link of `EB2` goes down and then up several times.

paraschos, 2016-10-21:
@marcoaccame yep, same stuff but maybe I was not spotting something. Thanks for taking a look. 

By the way I did **NOT** press the e-stop but I got that the boards EB7/8 thought that there was the fault button pressed. But maybe it was a random one. I am not sure if the `yarpRobotInterface` killed itself or I stopped it over the `yarp run`, i.e. stop from the moduleManager.

What about these types of warning?
>>>
2052	24.271603	WARNING	 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 146s 677m 60u: (code 0x0000000d, par16 0x01ba par64 0x0006001800710006) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
>>>

Not related with the EB communication errors but I am curious :)

julijenv, 2016-10-21:
Hi @paraschos ,
the link in between board EB2 and EB3 which are the board in the back (when looked from the back of the robot first picture- the one on the top left corner EB2 (see picture below - this board is located beneath another one but the connectors Im going to talk about later can be seen from my pciture and can be unconnected easily and check by you) and the second one EB3 is located at the right bottom corner).
On the Board EB2 you should check the cable labelled E2 attached to the connector P3 (that you can see on the picture, coinnector on the upper right corner) and the end of the cable E2 goes to the board EB3 on the connector P2, connector is on the upper left corner on the picture BUT watch out the board is upside down so that the P2 is on the bottom left corner.

[image]

[image]
[image]

check continuity in between every single wire from E2 from the both side of it,
Try to pull out the wires out from the connector, test if or not the wire is about to break or not. If it keeps in position without breaking it is good, if it breaks, just recrimps the cable and you will be done.
let us know

marcoaccame, 2016-10-21:
Hi @paraschos,

it tells that: the transmission phase of the control loop has lasted more than its assigned time. It lasted par16 = 0x01ba = 442 usec. There was a a burst of 6 CAN frames which the board had to transmit. The previous activity was with exec times of (RX = 0x0071 = 113 usec, DO = 0x0006 = 6 usec, TX = 0x0018 = 24 usec). 

You need to know that:
* The maximum allowed exec times are RX = 400 usec, DO = 300 usec, TX = 300 usec. Their total is 1 ms.
* In RX phase the ETH board decodes received UDP packets and received CAN frames.
* In DO phase it performs computations for motion control, collects strain, mais, skin, inertials data etc (if applicable).
* In TX phase the board blocks until all CAN frames or UDP packets to transmit are delivered to the HW transceivers.
* A CAN frame w/ payload of 8 bytes transmits in about 100 usec, hence a burst of frames may cause overflow. It is for this reason that the board logs inside par64 the number of CAN frames it had to transmit.
* Even if a limited number of consecutive phases overflow, the regularity of the control loop is immediately recovered because ... the start of each phase is controlled by a separate HW timer which eats up idle time to recover its pace.
* Consider that duration overflow is always logged and you seldom read such warnings.
* And even if you read some of these messages, nothing dangerous happens.

I hope I have satisfied your curiosity.

More about the internals of the control loop and its diagnostics I cannot explain unless we move to Q&A.

paraschos, 2016-10-24:
@julijenv Thx you for the description and the info!

Update: The E2 white cable at EB2/P3 connector was not making proper contact. Since we don't have the proper tools to re-clamp it we put a bit of solder and seams to made the trick. I have been running the robot for 1h and move it quite a bit (including pole, standing, yoga) and I didn't get any communication errors. We have a new problem with head j:0 and overcurrent protection. We will
try to figure out and if not, we will open another issue. 

@marcoaccame my curiosity is more than satisfied :) Ty

closed 2016-10-24 by paraschos