robotology/mesh-support #321: [icubBarcelona01] Impedance sensors malfunctioning?

jypuigbo, 2016-11-02:
Hi, 

### Description of the failure
Our robot has some ticks in the left hand, apparently due to some problem with the impedance sensors. The ticks are happening more often as the robot is on and it seems dangerous. See [video](https://drive.google.com/open?id=1yqEpWuPRqaUjRDSDK-ulw1a6fTik-mbjkA). 

### Detailed conditions and logs of the failure
We first saw it running `actionsRenderingEngine --motor::impedance on`, what showed the shaking and sporadic red arrows in the `icubGui` indicating imaginary forces in the hands. 
Then, we closed everythin and: 
- We run everything in  `icub_startup.xml` files
- We send the `calib all` rpc command to `/wholeBodyDynamics/rpc:i`
- We run `actionsRenderingEngine --motor::impedance on`

Then the shakes begin again but no arrows appear anymore in the `icubGui`. 

When we try to send `calib table` to the `/actionsRenderingEngine/cmd:io`, it stops after moving a few centimeters from the original position. 

Thanks for any insight on the reason of the issue. 
@randaz81 @pattacini @clement-moulin-frier @sockchinglow

randaz81, 2016-11-03:
Can you check all the connections, also in the core modules (e.g. between skinmanager and wholebody etc) and tell me if the skin is somehow involved?

jypuigbo, 2016-11-14:
Our apologies for the late reply. It effectively seems that it was related to the skin. We managed to calibrate successfully and it hasn't given any more problems to date. Thanks!

closed 2016-11-14 by jypuigbo