robotology/mesh-support #324: List of problems

amaroyo, 2016-11-08:
Ciao!

Please, take a look to this list of several issues regarding our reddy iCub!:

1. We are missing the two hangers on iCub's shoulder.

2. There are a lot of screws missing, mostly from the plastic covers of the legs, covers of the arms and the screws attaching the skin to the body.

3. Left shoulder cable is broken.

4. Left hand, ring and pinky fingers are not working - most likely the cable is broken.

5. Both covers of the hands are broken (fixing them with glue doesn't work anymore). The right hand is in critical status, the actual hand with the fingers and boards is getting deattached from the palm.

6. We are missing a silicon ring on the left thumb.

7. Right hand, middle finger there are tiny cables going out... A movement between those fingers can cut them.

Please, tell me if you need some more information! Hope to see you soon :)

reafrancesco, 2016-11-08:
I would add the 8th point:
8. thumb proximal of the right arm is not moving despite a correct reading from the encoder.

@amaroyo will help fixing the iCub even if I am not here.

julijenv, 2016-11-09:
Hi @amaroyo and @reafrancesco ,
I need you to be more specific for every single of your issues with your iCub:
  1. Do you have the main piece (see picture below) 
[image]
and you request just the little ring screw (uncircled in red on the picture above)
  2. Would it be possible to point out, on some pictures you can take of the leg cover, where are the missing screws, in order to fetch the good screws to replace on the robot.
  3. take a picture of the broken cable
  4. I really know the problem
  5. take a picture of what is broken I cant really understand what is actually ?!?!?
  6. easy peasy
  7. picture
  8. I guess the proximal joint is broken

From the list you sent, only point 6 is a critical one cause we need to oreder some of the pieces you mentionned (but still the pictures will give us more information).

Provide me with the pictures i ask uou take and we will jave a clearer ideas of what is going wrong and what i will neeed to bring you to repair most or the overall issues.

thx in advance

amaroyo, 2016-11-09:
1) The things that should be put in the holes on the back, see picture @reafrancesco if you can give some other detail
[image]

2) Missing screws
[image]
[image]
[image]
[image]

3) Broken cable - also stuck between the two metal joints 
[image]

5) Covers broken and deattached 
[image]
[image]

7) Mini cables
[image]

8) I think so, @reafrancesco can confirm.

Thanks!

julijenv, 2016-11-11:
Hi @reafrancesco and @amaroyo ,
Let's talk more precisely from every single issue to understand what I can possibly do and what you will need to order or get in contact with Fabrizio L. (I will write down a mail to you to let you know what part you will need to order).
   1. To hang the robot It looks like you do not have the claw system, right? from the picture I can see anything else than the boards. (if you dont have the claw, we will need to get it before being able to mount it.
  2. The leg cover has lost screws not only because of the manutention team couldnt find it but because actually the cavoer is broken in all the spot where the screws are missing. ( I guess that either the cover got older or the CODYCO-KOROIBOT team manage to break them during their experiments. I may suggest you to get new cover or foget about making it repaired.
For the other screws of the arms cover I will make a list and fetch tehm to be able to remount them on your robot next time.
  3. cable 8 pinched and for sure will need to replaced.
  4. we will see on the spot (test mechanical or electrical)
  5. Right dorso cover broken (we may have a spare one around or you will have to buy it). Left palm unglued - fixable.
  6. remount silicon tube
  7. Nothing to worry about, we will brush them along the finger.
  8. To be checked, rewiring expected.