robotology/mesh-support #326: Neck tendons broken

DanielePucci, 2016-11-08:
Please **fill in the following two sections** and then **remove any comment and/or remark become unnecessary**:

### Description of the failure

Neck tendon broken

### Detailed conditions and logs of the failure

While testing new control algorithms for balancing, the robot fell and the shock on the head made the tendon break.

@julijenv @spada1978 

---
**Remark**: if you are not a daily icub-support user, please read carefully the guidelines for contributing linked at the top of this page. In particular, the **name of your robot** must be always indicated: either using **labels** if you're already a contributor, as per standard, or by specifying it while detailing the issue.

DanielePucci, 2016-11-08:
@francesco-romano @S-Dafarra

spada1978, 2016-11-09:
replaced the cable B IIT code 5964 and performs the wiring again

closed 2016-11-09 by spada1978