robotology/mesh-support #328: Error during the calibration of the Left_Arm II

misaki43, 2016-11-16:
Hello,

This morning we launched the robot and the issue on the left/right arm came back (see #308).
The first time we launched it, the two arms weren't calibrated well (see [log_pc104_yarprobotinterface_3199.txt](https://github.com/robotology/icub-support/files/594487/log_pc104_yarprobotinterface_3199.txt)).
The second time, only the left-arm wasn't well calibrated (see log [log_pc104_yarprobotinterface_3428.txt](https://github.com/robotology/icub-support/files/594488/log_pc104_yarprobotinterface_3428.txt))
Here are the values of the left-arm joint 0-4 when we launch yarprobotinterface the second time:
[image]

Note: I launched yarprobotinterface from robot-config, where we had corrected the files. I didn't use the robot since we had done the fine calibration.

julijenv, 2016-11-16:
Ciao @misaki43 ,
Could you make an easy test with the robto.
Coujld you try to to launch the robot as if you wanted to calibrate it from scratch RED BUTTON PRESSED. meaning you put the robot in standing mode.
Check the values from yarpmotorgui - tey shoud be all close to zero (+-10, I guess). If not try to movef a bit every joint of the shoulder for about +-5 degree  and check  I joint 2 is way furthe if the values jump at some point.
If everything is ok - values in the limit of acceptability. Put back the robot in the almost end of calibration after starting yarproobtInterface and see if the robot start correctly.
Let us know

misaki43, 2016-11-17:
Hello @julijenv,
The values of the joints 2 were far away from +-10:
left arm: j1-j8 : 0.192, 2.131, -216.9, 8.43, -93.1
right arm: j1-j8: 10.6, 0.8, -103.6, 10.4, -91.3
There was no value jump when we move the joint2 and when we moved it the values varied as following:
left arm from -64.5 to -296.2  (diff : 231,5) for an angle < 180°
right arm from -195.4 to 88.3 (diff : 283) for an angle <= 180°

julijenv, 2016-11-22:
Hi @misaki43 ,
Did you try to get those values by putting the robot in the standing configuration and afterwards turning ON the switch of the motors ???(unfortunately the robot need the boards to reboot in order to get the good values).
Give me a call on skype when you will retry
 C u soon

dogoepp, 2016-11-29:
I do not remember how we did but I'll try to make a few attempts today and report the results.

The board to remove, are they those that the Motor switch powers on ? What is "good values" referring to ?

misaki43, 2016-11-30:
Hello,

we noticed that different configurations were used depending on the current folder from which `yarprobotinterface` was started. Each led to a different behavior at startup (different angles for joints 0-4).

When starting from within `robots-configuration/iCubNancy01`, the joint angles for the standing pose (arms along the body) were

Left arm

| trial time | j0  | j1  | j2  | j3   | j4  |
|------------|-----|-----|-----|------|-----|
| 10:45      | 6.8 | 3.3 | 2.8 | 10.7 | -90 |
| 11:03      | 6.7 | 3.3 | 2.7 | 10.7 | -90 |
| 11:10      | 6.7 | 3.4 | 2.8 | 10.7 | -90 |

Right arm

| trial time | j0  | j1  | j2   | j3  | j4  |
|------------|-----|-----|------|-----|-----|
| 10:45      | 5.2 | 6.6 | -5.7 | 8.4 | -90 |
| 11:03      | 5.2 | 6.6 | -5.7 | 8.4 | -90 |
| 11:10      | 5.2 | 6.6 | -5.7 | 8.4 | -90 |

Then we understood that it was a mistake during the switch from icub-main to robots-configuration. We just had not done `make install` for the configuration files. Before doing it, we backed up the installed configuration in `$ICUBcontrib_DIR/` However, as soon as we installed the updated configuration, `yarprobotinterface` would systematically lead to a segmentation fault which cause seems to be

```
[ERROR]cfw2can [0] thread:0 msg:91 joint:0 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version
[ERROR]cfw2can [0] thread:0 msg:91 joint:1 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version
[INFO]
[INFO]**********************************
[INFO]iCubInterface CAN protocol: 1.6
[INFO]Firmware report:
[WARNING]cfw2can [0] joint: 0 can_address:  1 Unable to detect firmware version. Old firmware running?
[WARNING]cfw2can [0] joint: 1 can_address:  1 Unable to detect firmware version. Old firmware running?
[INFO]cfw2can [0] joint: 2 can_address:  2 board type: 3 (4DC) version: 2.15 build: 87 CAN_protocol:1.6
[INFO]cfw2can [0] joint: 3 can_address:  2 board type: 3 (4DC) version: 2.15 build: 87 CAN_protocol:1.6
[INFO]cfw2can [0] joint: 4 can_address:  3 board type: 3 (4DC) version: 2.15 build: 87 CAN_protocol:1.6
[INFO]cfw2can [0] joint: 5 can_address:  3 board type: 3 (4DC) version: 2.15 build: 87 CAN_protocol:1.6
[INFO]**********************************
[INFO]
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]  It has been detected that your control boards are not using the same
[ERROR]  CAN protocol used by iCubinterface. iCubInterface cannot continue.
[ERROR]  Please update your system (iCubInterface and/or your control board firmware.
[ERROR]  For further information please visit: http://wiki.icub.org/wiki/Firmware
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]checkFirmwareVersions() failed. CanBusMotionControl::open returning false,
```

This error is raised whatever the folder from which `yarprobotinterface` is started.

A full log is attached to this message.
[30-11-13h07-firmwarePblm.txt](https://github.com/robotology/icub-support/files/621773/30-11-13h07-firmwarePblm.txt)

Thanks!

@olivierrochel-inria @serena-ivaldi @jbmouret

randaz81, 2016-11-30:
> However, as soon as we installed the updated configuration, yarprobotinterface would systematically lead to a segmentation fault

There is no correlation between the use **robots-configuration** repository and the error shown above. The system seems to be unable to communicate with the board with can address 1 on can line 0. Use the `canloader` tool to check if the board is properly connected to the can bus. @julijenv

dogoepp, 2016-11-30:
Thanks for the notice, we'll try hopefully tomorrow. 

I'm nonetheless confused since I cannot remember of any other manipulation other than on the configuration files before this error arised.

dogoepp, 2016-12-01:
Excuse-me @randaz81, I don't know how to do the checking with canLoader. Could you point me to a documentation for this ?

Thanks

julijenv, 2016-12-01:
Hi @misaki43 , @dogoepp ,
from the excel table that you post I guess that everything looks alright (close to zero) so try to launch again the robot from that specific repository (after having debug the issue that I will be talking in few seconds) and I guess the robot should calibrate correctly.
If the robot dont stat it is becasue of this message :
[WARNING]cfw2can [0] joint: 0 can_address:  1 Unable to detect firmware version. Old firmware running?
[WARNING]cfw2can [0] joint: 1 can_address:  1 Unable to detect firmware version. Old firmware running?
Could you please open a terminal on pc104 (-X) and laucnh canLoader on net0 - cfw2
I guess you wont be able to see ID 1 BLL board.
At first, try to see if you power cycle the boards (switch OFF and ON the motors) and connect on the can0 within 5 sec BLL board show up ( if yes the FW needs to be reflash on the BLL board).
If you still can NOT see it even in bootloader mode (in the first 5 sec of the board getting powered) you may have to see if there are some problem with the connection net0 at the level of the neck or at the level of the head.
LEt me nkow your findings.

julijenv, 2016-12-02:
As far as understood, the calibration went alright.
The robot seems now behaving correctly.

closed 2016-12-02 by julijenv