robotology/mesh-support #329: Encoder spikes on left wrist of iCubGenova04

traversaro, 2016-11-17:
### Description of the failure
The robot was just started. 

### Detailed conditions and logs of the failure
The output of yarprobotinterface is : 
~~~
[WARNING] from BOARD 10.0.1.24 (leftArm-j4_7), src LOCAL, adr 0, time 612s 183m 833u: (code 0x0200000a, par16 0x0002 par64 0x0000000000000033) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
[WARNING] from BOARD 10.0.1.24 (leftArm-j4_7), src LOCAL, adr 0, time 613s 183m 830u: (code 0x0200000a, par16 0x0002 par64 0x0000000000000016) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
~~~

traversaro, 2016-12-01:
cc @iron76

iron76, 2016-12-01:
@randaz81 who should this issue be assigned to?

iron76, 2016-12-07:
@randaz81 we are thinking about using the iCubGenova04 robot at the winter school and therefore we need to solve this issue. Any update? @traversaro

traversaro, 2016-12-07:
The encoder spikes have been solved, Julien is now working on some problems on the right arm.

julijenv, 2016-12-07:
has been solved

closed 2016-12-07 by julijenv