robotology/mesh-support #334: Unexplained crash and unresponding CAN boards

dogoepp, 2016-12-06:
## Context

After resolution of issue #328, we observed that `yarpmanager` was no longer able to start nodes located on icubsrv. We went to investigate what probably is a small issue when the following happened.

## Events

I left the robot on during our lunchtime (`yarprobotinterface` started). I don't remember if any program other than `yarprobotinterface` was running but we did not start any program that was meant to give orders to move. I thought that Serena had told me that the robot can be left on for long stretches of time.

When we came back, I unlocked icubsrv and icub01. They had not gone to sleep (it is disabled on both computers). It seemed that PC104 had frozen, as the excerpt from the SSH session bellow shows. I pressed ^C, as I was not sure what the messages meant and just thought that there was an error in `yarprobotinterface` and wanted to restart it. Then the robot's torso tilted to the front, reaching its joint limit. When trying to raise it, it resisted. It indeed seemed that the position control was still active as the arms did not fall (the right usually does, but not the left). I then pressed the emergency stop and we put the robot to a resting position.

No SSH session could any-more be opened to PC104 so we power cycled it. I suggested that we try to start again `yarprobotinterface` (I don't know if it was a bad idea), and we got error messages about CAN boards, as in issue #328. See the log file `pc104-ssh-session-qfter-crash.txt` for more on these errors.

I did not dare to do any other attempt and switched everything off.

## PC104 crash log excerpts

Excerpt from SSH session on PC104 (file `pc104-ssh-session.txt`):

```
[DEBUG]cfw2can [0] printing boards infos:
None
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[DEBUG]yarprobotinterface running happily 
[DEBUG]cfw2can [0] thread ran 596 times, req.dT:10[ms], av.dT:10.06[ms] av.loopT :0.02[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]cfw2can [0] printing boards infos:
None
[DEBUG]cfw2can [0] thread ran 595 times, req.dT:10[ms], av.dT:10.07[ms] av.loopT :0.02[ms]
yarp: No connection to nameserver
yarp: *** try running: yarp detect ***
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
yarp: No connection to nameserver
yarp: *** try running: yarp detect ***
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[DEBUG]yarprobotinterface running happily 

Message from syslogd@pc104 at Dec  6 12:31:31 ...
 kernel:[ 7534.056207] Stack:

Message from syslogd@pc104 at Dec  6 12:31:31 ...
 kernel:[ 7534.056225] Call Trace:

Message from syslogd@pc104 at Dec  6 12:31:31 ...
 kernel:[ 7534.056227]  <IRQ> 
```

Excerpt from `syslog`:

```
Dec  6 12:31:31 pc104 kernel: [ 7534.056043] INFO: rcu_preempt detected stall on CPU 0 (t=15000 jiffies)
Dec  6 12:31:31 pc104 kernel: [ 7534.056048] sending NMI to all CPUs:
Dec  6 12:31:31 pc104 kernel: [ 7534.056054] NMI backtrace for cpu 0
Dec  6 12:31:31 pc104 kernel: [ 7534.056057] CPU 0 
Dec  6 12:31:31 pc104 kernel: [ 7534.056059] Modules linked in: aerfx2(O) snd_pcm_oss snd_mixer_oss cfw002(O) nfsd nfs nfs_acl auth_rpcgss fscache lockd sunrpc snd_intel8x0 snd_ac97_codec snd_pcm snd_page_alloc snd_seq snd_seq_device snd_timer parport_pc parport acpi_cpufreq mperf evdev i915 cdc_acm snd video drm_kms_helper drm iTCO_wdt i2c_algo_bit processor iTCO_vendor_support button serio_raw i2c_i801 coretemp rng_core i2c_core pcspkr soundcore ac97_bus ext4 crc16 jbd2 mbcache ohci_hcd squashfs loop aufs(C) nls_utf8 nls_cp437 vfat fat sg sd_mod crc_t10dif microcode usb_storage ata_generic e1000e fan thermal e100 thermal_sys firewire_ohci ata_piix firewire_core uhci_hcd mii ehci_hcd crc_itu_t libata usbcore scsi_mod usb_common [last unloaded: scsi_wait_scan]
Dec  6 12:31:31 pc104 kernel: [ 7534.056144] 
Dec  6 12:31:31 pc104 kernel: [ 7534.056148] Pid: 0, comm: swapper/0 Tainted: G         C O 3.2.0-4+icub.1-rt-amd64 #1 Debian 3.2.57-3 PhoenixAward 945GM/945GM
Dec  6 12:31:31 pc104 kernel: [ 7534.056154] RIP: 0010:[<ffffffff811c0d60>]  [<ffffffff811c0d60>] __delay+0x7/0x9
Dec  6 12:31:31 pc104 kernel: [ 7534.056165] RSP: 0018:ffff88007e403df0  EFLAGS: 00000847
Dec  6 12:31:31 pc104 kernel: [ 7534.056168] RAX: ffffffff811c0d97 RBX: 0000000000002710 RCX: 0000000000000040
Dec  6 12:31:31 pc104 kernel: [ 7534.056172] RDX: 000000000020b60f RSI: 0000000000000200 RDI: 000000000020b610
Dec  6 12:31:31 pc104 kernel: [ 7534.056175] RBP: ffff88007e40f020 R08: 0000000000000002 R09: 0000000000000000
Dec  6 12:31:31 pc104 kernel: [ 7534.056178] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffff81623b80
Dec  6 12:31:31 pc104 kernel: [ 7534.056182] R13: 0000000000003a98 R14: ffff88007e40ec50 R15: ffffffff81071d85
Dec  6 12:31:31 pc104 kernel: [ 7534.056186] FS:  0000000000000000(0000) GS:ffff88007e400000(0000) knlGS:0000000000000000
Dec  6 12:31:31 pc104 kernel: [ 7534.056190] CS:  0010 DS: 0000 ES: 0000 CR0: 000000008005003b
```

## Complete log files

[pc104-ssh-session.txt](https://github.com/robotology/icub-support/files/634176/pc104-ssh-session.txt)  
[pc104-ssh-session-after-crash.txt](https://github.com/robotology/icub-support/files/634177/pc104-ssh-session-after-crash.txt)  
[system-logs.zip](https://github.com/robotology/icub-support/files/634178/system-logs.zip)

## Interested persons
@misaki43 @serena-ivaldi @jbmouret @olivierrochel-inria

serena-ivaldi, 2016-12-06:
It seems yarpserver and the yarp modules on the machines icub01 or icubsrv were closed when the computers went to sleep mode.

dogoepp, 2016-12-06:
As I said, the sleep mode is disabled on these computers. How could this happen ?

serena-ivaldi, 2016-12-06:
Try to do this experiment: just launch yarpserver, yarmanager and maybe icub_sim, then try to repeat the error by letting the computer go in locked mode.

dogoepp, 2016-12-07:
(sorry, I forgot to add the tag)

We started the robot this morning. No more error messages about the can boards. Also, I locked both icubsrv and icub01 without a problem. Admitedly, the time during which the computers were locked was less than a minute.

This has to be watched but the issue as such could be considered as solved

closed 2016-12-07 by dogoepp