robotology/mesh-support #335: Issues with the right hand thumb (oppose and proximal)

AlexAntn, 2016-12-09:
### Description of the failure

Right thumb proximal is moving to extreme positions when trying to control thumb oppose (100+ degrees angle).

### Detailed conditions and logs of the failure

While running some experiments we broke the right thumb oppose cable. We successfully replaced the cable, but upon starting up the finger was not starting up correctly (it was trying to reach the 100+ angle).

Suspecting a calibration issue, we followed the guidelines provided on #195 to calibrate the finger. 

Upon testing the thumb oppose with yarpmotorgui, we saw that the direction of movement was inverted (when instructing to go to 0, the joint was moving to 90). Again following the guidelines above, we inverted the controller for the thumb oppose, and it now moves accordingly.

The thumb proximal, however, still behaves in a weird way:

- When trying to control joint 9 (proximal) on its own, it won't move at all.
- When trying to control joint 8 (oppose). joint 9 will jump to the extreme position and exerting torque (100+ angle).

We also tried inverting the controller for the thumb proximal (joint 9) but it made no difference in this behaviour.

The proximal also behaved in a weird fashion upon start-up:

- With the default control values, it kept oscillating (sometimes around the extreme position 90+ angle, other times between 40 and 70 degrees)
- With the inverted controller values (for kp, kd, and ki) we have the correct startup (joint 8 at 45 degrees, joint 9 at 0 degrees), but then we get the behaviour described above when we try to control joint 8 and 9.

Any ideas into what the problem might be, or what else we can check to debug this issue?

julijenv, 2016-12-19:
First thing first,
you may understand that unfortunately j8-j9-10 of the V2 forearm are kind of accoupled so if one of them is not properly functionning the movement is not garanted.
First make sure the FC connector on the top of the MAIS board - at the level of the knuckles of the hand- is perfectly plugged-in, if not replugged it and relaunch the robot.
Second attempt, Please check when motor idles that j9 can be moved (use a flat screwdriver at the level of the elbow to spin the middle motor screw - prox thumb) and that the encoder values displayed linear values form extend to close position.
I am behind icub.support acocount on skype give me a call if ever needed.

AlexAntn, 2017-01-06:
Hello @julijenv , happy new year!

Sorry I haven't looked into this for a while, as i was not in plymouth these days.

I have checked both options as you described:

1) the board seemed to be slightly out of place, so i plugged it back in and started the robot. The behaviour is still the same however...

2) I have checked the encoders for both joint 8 and joint 9, and both give linear and correct outputs.

This error seems to be at the level of the controllers, since the encoders are returning the correct values in the different positions, but i am unable to move the fingers in the yarpmotorgui (joint 9 immediatelly jumps to 100 Degrees when turned on, and I can't change the angle on joint 8).

Will you be available for a skype call on monday (9th January)?

Thanks for the help!

julijenv, 2017-01-09:
Hi @AlexAntn ,
I'm right back my computer on the icub-support account on skype.
If vever needed just raise an hand.

AlexAntn, 2017-01-10:
Hello @julijenv , thanks for the skype meeting!

On the skype meeting we agreed the problem might be on the wiring. We took some pictures of the wiring (as much as possible; Taking pictures of the motor itself would require removing the cable of the finger, which we have yet to do since it would require replacing the cable).

Here are the pictures (with captions):

[image]
joint 8 cable attachment

[image]
Cables going into the motor plate (showing the cable covers connecting properly to the small aluminum covering board)

[image]
[image]
Two attempts at taking a picture of the motor (first picture we can see a tiny bit of the cable with the finger closed; second picture with the finger opened, no cable present)

[image]
[image]
Manually closed the finger to the extreme position (picture 1), then tried opening it with the screwdriver on the motor to the other extreme (picture 2)

[image]
[image]
Manually opened the finger to the extreme position (moving also joint 9, picture 1), then tried closing it to the extreme position with the screwdriver on the motor (picture 2)

I am sorry there are so many pictures, we tried to be as thorough as possible without removing anything yet. If you think it is best to remove the cable from the finger and take pictures of the motor, let us know!

As for the next skype meeting, I would suggest we do it either tomorrow (though that might be too soon), or next week.

Thanks in advance!

julijenv, 2017-01-10:
Hi @AlexAntn ,
from the picture 3 and 4 it is obvious that the problem is mechanical, you forgot to unroll the wire for one turn around the pulley for either the cable going in than the cable gettin out.
I will give you an hand to rewirie it whenever you are availble. It ouwld nice if you could have someone to give you physically an hand in Plymouth to do the wiring.
Let me know,

AlexAntn, 2017-01-16:
Hello @julijenv ,

Upon discussing with the local technician, we would be available to do the rewiring on thursday or friday, if it suits you. Alternatively, we can do it early next week if it is best for you!

Thanks!

julijenv, 2017-01-16:
Hello @AlexAntn ,
To me it looks like more than possible at those days so just reminds me it on the day before it, whne your technician will be here for sure and possibly the time in the day,
C U soon,

AlexAntn, 2017-01-18:
Hello @julijenv , we will rewire the thumb tomorrow morning, I will contact you on skype when we are in the lab.

It should be around 10 a.m. (11 a.m. in Italy).

Thanks!

julijenv, 2017-01-18:
all right

julijenv, 2017-01-19:
@AlexAntn , just a little overview to how to rewire this joint at the level of the motor pulley.
[image]

julijenv, 2017-01-19:
[image]
[image]

AlexAntn, 2017-01-19:
canLoader pictures:

[image]
NET2

[image]
NET 6

[image]
NET8

AlexAntn, 2017-01-19:
Hello @julijenv ,

Thanks for the time and patience you had with us! I'm sending the output of the robotInterface here:
[Log.txt](https://github.com/robotology/icub-support/files/717527/Log.txt)

julijenv, 2017-01-23:
Hi @AlexAntn ,
Fron the log file there is something aakward with the MAIS, could you try to to remove the dorso cover and see if all little boards on the longest board about the knuckles are all paralels and perfectly plugged, especially the one the most close to the thumb.
in any case Im behind my desk during the day, so let me know if you want me to help you.

AlexAntn, 2017-01-23:
Hello @julijenv ,

We have opened the hand and checked the boards. They seem to be fine (pictures here), and we tried to turn the robot on again but nothing changed (we thought maybe the cover was putting pressure on the boards, so we tried with the plastic cover removed).

Here are the pictures of the boards:

Above:
[image]

Thumb side:
[image]

Index finger side:
[image]

from the front:
[image]

Thanks for the help!

julijenv, 2017-01-23:
Hi @AlexAntn ,
could you one more time try to detect if MAIS board can be seen on net 6???
from the log file it looks like a lot of error come from it.
let me know

AlexAntn, 2017-01-23:
Hello @julijenv ,

We have tried fixing the PID values for joint 9 (like we did for joint 8, looking into the PID values for the left hand), and it works! I think the reaction time might be a bit slow, but it is moving correctly now!

As for the wrist, we have looked both on the small board at the wrist level and at the 2B2 board on the arm. The cables all looked fine, and were solid upon testing (on the 2B2 board). We took some pictures:

2 pictures of the wrist-board:

[image]
[image]

picture of 2M2 cable on board 2B2:

[image]

picture of 2M1 cable on board 2B2:

[image]

julijenv, 2017-01-24:
Hi @AlexAntn ,
could you please check if the power cable is perfectly plugged into the board 2B2 (the rainbow colour 8-pin connector).
could you please try to take a picture of the top part of the board as well as the left and right part, I need to see every single connector and wire .
thx in advance,
At the level of the wirst everything looks alright to me.

AlexAntn, 2017-01-25:
Hello @julijenv ,

I took some pictures of the board:

Top-back of the board 2b2:
[image]

Top-front of the board 2b2:
[image]

Left of the board 2b2:
[image]

Detail on (slightly damaged?) connector on left of board 2b2:
[image]

right side of the board 2b2:
[image]

back of board 2b2:
[image]

julijenv, 2017-01-26:
hi @AlexAntn ,
ok for me it is not clear whether the real pblm is coming from the cable which blossom a bit. try to recrimp it, and let me know if it solve the issue but a i doubt about it.
Did you try to pull out everything single wire inside every single connector for 2M1 and 2M2 to see it one is hanging but not really attach.
let me know

AlexAntn, 2017-02-15:
After some further inspection on the 2B2 board, it was detected that some metal from the drilling of the elbow screw got into the board. Upon clearing it, the robot works perfectly!

Thanks @julijenv for the help in fixing our robot!

Closed

closed 2017-02-15 by AlexAntn