robotology/mesh-support #338: Overcurrent on joint 1 of left leg

valegagge, 2016-12-16:
When I switch on pwm on joint 1 of left leg, it goes in overcurrent fault.

julijenv, 2016-12-21:
I'd already check last week if there was an electrical problem but couldnt find it, Could you please make some test to narrow down the reason fo this overcurrent?

iron76, 2016-12-22:
@valegagge and @julijenv any news in this concern?

julijenv, 2017-01-02:
Hi @valegagge,
The overcurrent was due to 2 phases of the motor touching each other. One random metallic fall inside the 2FOC board leading to link in between the 2 phases.
Everythin' has been replaced on the leg, please check if the message is gone on yarpprobotinterface.
let me know and close the issue if necessary

traversaro, 2017-01-05:
@valegagge The robots starts and the joint goes in "Not Configured". Perhaps it was commented out in some configuration file?

traversaro, 2017-01-05:
[image]

traversaro, 2017-01-05:
We enable the joint in calibration and everything works fine, closing the issue.

closed 2017-01-05 by traversaro