robotology/mesh-support #339: rotate encoders of fingers abduction joints of both hands

valegagge, 2016-12-16:
@julijenv try to move encoders, but they have been glued

julijenv, 2016-12-21:
Hi @valegagge ,
we managed (Aesandor Spadoni and I) to detached both arm and proceed to the enlarge the seating place of the magnet. You should be able to calibrate the fingers abduction now.
LEt us know, lcse the issue ASAP yuo testify that the issue is over.

julijenv, 2017-01-02:
Hi @valegagge ,
Both cover of the upper arm are still waiting to get remounted, please let us know when you assert that all motors are working fine and will come to close them.
let us knojw,

traversaro, 2017-01-04:
I tested the robot and it seems there are some problems on the head and the left arm (several joints start in hardware fault or idle) with this error messages:
~~~
211753	171,369556	ERROR	HasSpeedEncoder  parameter not found
211755	171,369571	ERROR	 from BOARD 10.0.1.1 (leftArm-j0_3), src LOCAL, adr 0, time 49s 623m 75u: (code 0x02000003, par16 0x0003 par64 0x0000000000000000) -> MC: 2FOC hall sensors fault. Invalid sequence in motor Hall effect sensors, please check motor hall cable connections. par16 = ID of joint. + .
211784	171,440254	ERROR	 from BOARD 10.0.1.24 (leftArm-j4_7), src LOCAL, adr 0, time 49s 749m 337u: (code 0x05000016, par16 0x0402 par64 0x0000000000100130) -> CFG: EOtheEncodeerReader cannot be configured. verification fails. in p16 the number of joints (msb) and failure mask (lsb), in p64 for each physical joint a byte with (port, errorcode) + .
211790	171,621967	ERROR	HasSpeedEncoder  parameter not found
211832	171,863872	ERROR	HasSpeedEncoder  parameter not found
211932	172,368410	ERROR	HasSpeedEncoder  parameter not found
211971	172,610700	ERROR	HasSpeedEncoder  parameter not found
212013	172,842142	ERROR	HasSpeedEncoder  parameter not found
212224	176,371892	ERROR	 from BOARD 10.0.1.1 (leftArm-j0_3), src LOCAL, adr 0, time 54s 623m 79u: (code 0x02000003, par16 0x0003 par64 0x0000000000000000) -> MC: 2FOC hall sensors fault. Invalid sequence in motor Hall effect sensors, please check motor hall cable connections. par16 = ID of joint. + .
212569	180,650815	ERROR	 from BOARD 10.0.1.24 (leftArm-j4_7), src CAN1, adr 0, time 58s 964m 107u: (code 0x00000019, par16 0x0401 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
212573	180,660830	ERROR	 from BOARD 10.0.1.24 (leftArm-j4_7), src CAN1, adr 0, time 58s 967m 107u: (code 0x00000019, par16 0x0401 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
212577	180,660856	ERROR	 from BOARD 10.0.1.24 (leftArm-j4_7), src CAN1, adr 0, time 58s 970m 107u: (code 0x00000019, par16 0x0401 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
~~~

I see that some repo of the robot are still on the `rtcmc3` (such as https://github.com/robotology/robots-configuration/tree/rtcmc3) so I don't know if it is actually a problem due to incompatible YARP/ICUB/firmware.

traversaro, 2017-01-05:
The same error actually persist after changing the robots-configuration branch to `master`.

traversaro, 2017-01-05:
The current status of the robot is the following: 
[image]

traversaro, 2017-01-05:
I will open separate issues for every problem.

traversaro, 2017-01-05:
Regarding joint 7, they are disabled in the calibration file until @valegagge provide the values for the calibrator.

traversaro, 2017-01-09:
This is the only blocking issue remaining on the robot.

closed 2017-01-10 by traversaro