robotology/mesh-support #340: cannot find icub_right_arm.xml and icub_left_arm.xml on iCubDarmastadt01

traversaro, 2016-12-16:
@giuliavezzani commented on [Fri Dec 16 2016](https://github.com/robotology/yarp/issues/1029)

Hi all, 

I'm looking for the following files:

https://github.com/robotology/robots-configuration/tree/master/iCubGenova01/hardware/motorControl

but for the `iCubDarmstadt01`, in order to perform [**arm fine calibration**](http://wiki.icub.org/wiki/ArmFineCalibration) and update the **zeros** values.

I couldn't find any files like those and, I didn't find even any file where the **zeros** for the robot are set.

@paraschos suggested to ask @traversaro for any suggestions.

Thanks,
Giulia

---

@traversaro commented on [Fri Dec 16 2016](https://github.com/robotology/yarp/issues/1029#issuecomment-267566879)

I think we can move this issue to `icub-support`, right?

traversaro, 2016-12-16:
`iCubGenova01` is a  iCub where most electronics is connected with the head computed with `CAN` networks, while `iCubDarmastadt01` is an `ETH` based robot, this is the reason while the configuration files are different. 

Unfortunately the documentation on the arm fine calibration is currently lacking (see https://github.com/robotology/icub-main/issues/240). As pointed out in https://github.com/robotology/icub-main/commit/82b56c63df0a99b8cd2246142c1d70e9fc69516d#commitcomment-14458176 , the files that need to be modified in `ETH` robots are the `CalibrationDelta` contained (for example) in https://github.com/robotology/robots-configuration/blob/master/iCubDarmstadt01/calibrators/left_arm-calib.xml . Check the history of those files to see example of commits updating the fine calibration. 

**Caveat**: if you need to update the fine calibration of the neck, please check out https://github.com/robotology/icub-main/issues/388 , that affects also `iCubDarmstadt01`.

giuliavezzani, 2016-12-16:
Thank you so much @traversaro! 
I just need to calibrate the arms.

giuliavezzani, 2016-12-16:
Thanks @traversaro again, I managed to calibrate the arms :tada:

closed 2016-12-16 by giuliavezzani