robotology/mesh-support #343: 1B2 board (left arm) not showing up

kt10aan, 2016-12-20:
### Description of the failure
The 1B2 board on the left arm is not showing up

### Detailed conditions and logs of the failure
After replacing a couple of the left shoulder tendons, we tested the robot and the arm was working properly.

But after a couple of shutdown/rebooting the robot, the 1B2 robot does not show up (Cannot read firmware version  and ABS ENCODER ERROR)

I inspected the boards visually and all the connector (on the board 1B2 itself and all the way to the boards on the back of the robot) seem fine. Also, all the lights seem to be the same on all three boards of the left arm.

I also tried to boot the boards by connecting to CAN Loader (CAN1) within a couple of seconds of switching on the motors). The board would still not show up. As it doesn't show up I have no way to reflash the firmware (as in the past, sometimes that would fix similar errors).

I also checked the shoulder sensor connection and it seems fine.

What further tests/debugging should I do?

julijenv, 2016-12-21:
Hi @kt10aan ,
Please consider to send always a log file attached to the failure in order for us to debug instead of randomly giving u hint on every possible ways the robot could have been damaged ;P.
If you have seen a message like cannot read firmware version you have surely a problem of connection with the specific board if you look at it and the lights turn ON the same way. Or the 4 pin connector has an issue ( try to see if you have continuity. or one of the 8 pin connecotr attached to 1B2 has one wire broken/unplugged. I would try this first. When the connection with the board is retalished, see if the ABS ECNODER ERROR still appear. if yes sen me  link to the log file.
tjhx

julijenv, 2017-02-02:
Hi @kt10aan , any news???

kt10aan, 2017-07-12:
Sorry for the super delayed response. We were using just the right arm the past months :)

So, I came back to the issue, inspected everything, and I found broken cables at one connector in the back of the robot. See attached picture : (the arrow shows where the connector was, the actual connector is not showing in the photo)

[image]

I have two questions:
1) Is this connector the one that goes to 1B2?
2) Except the rightmost (where right is the right side of the icub) black cable (thicker than the other ones), all other cables were broken. Is their order the same as the other similar connectors? (From right to left: Black thick, black thin, white, green

julijenv, 2017-07-13:
Hi @kt10aan ,
please take a picture of the connector which you have in your hand (the broken one) and I will let you know how to recrimp everything back in place.
let me know.
However how many ways is this connector?

kt10aan, 2017-07-18:
The connector has six pins. Three of the cables were broken (except the big black one). But the three thin cables were too short to recrimp it so we made another connector, with new cables, to extend it and connect them to the original cables. (We checked all for continuity). We didn't have white and green cable so we had to improvise :)

Here is the photo of the connector as it is seen the way it goes into its socket  (right sid eof the picture is the right side of the icub, when we look from the back)

[image]

As I said above, original order was, from right to left: Black thick, black thin, white, green, empty, empty). Only the black thick cable was still intact, the order of the others I deduced it by looking at other connectors. Which means, it might not be absolutely correct.

In any case, it would be immensely useful for us to arrange a skype meeting this week if possible :)

julijenv, 2017-07-19:
Hi @kt10aan ,
please plug the connector into the 1B2, turn on the motor, wait 10s and use the canLoader to check the lnet1 with cfw2 and let me know if you got:
[image]
dont pay attention of the FT virtual sensor.
let me know,

kt10aan, 2017-08-22:
Hi @julijenv,

we moved to  new building so iCub + accessories was packed for some time.

So, we made a replacement connector and plugged it in 1B1. CanLoader still doesn't list the 1B2 board.

I have checked all the connections on 1B2 and everything seems to be fine.
I have checked all the connections from 1B2 to 1B1 and I cannot see any broken cables.
I also changed the filter in the N1 line that goes from 1B1 to 1B2.

Still, the board doesn't show up.

Could you confirm the order of the cables on the connector on 1B1 (as we look the back of the iCub, the one on the top of 1B1, that is closer to us (meaning there is another connector on the back of 1B1 at the same place).

As we had it now, we have " empty empty green white black-thin black-thick". Is this te correct order?

Is there anything else we should check?

julijenv, 2017-08-22:
Hi @kt10aan ,
As you shown from your first post the connector should be plugged into the connector point by the yellow arrow.
However the scheme of the cabling should be the same as the connector plug into the BLL in the other side of the board.
Strangely enough yours is the opposite?!?!? I would recommend you to do not plug the little black one as you named it - it looks like that it could make a short with the CAN_VCC
[image]
[image]
Your connector got plugged into P3 - the connextor in the other side of the board facing P3 is P10 - which already have the N1 cable plugged into it.

Just copy the cabling of the connector and you should be ok.
Let me know

kt10aan, 2017-08-24:
@julijenv 

I tried with the order you suggested and the 1B2 board shows up on canLoader :) Thanks!

Upon starting robotInterface though, there are errors, related to the HALL sensor.
Interestingly, the same issue happens to the right arm also.

I have checked most connections I could think of, and they seem fine (also checking for continuity).

I attach the logs here, one with the left arm active and the right arm disabled and the other one with the right arm active and the left arm disabled.

[OsakaLogRightArm.txt](https://github.com/robotology/icub-support/files/1247936/OsakaLogRightArm.txt)
[OsakaLogLeftArm.txt](https://github.com/robotology/icub-support/files/1247937/OsakaLogLeftArm.txt)

kt10aan, 2017-08-24:
@julijenv 

I tried with the order you suggested and the 1B2 board shows up on canLoader :) Thanks!

Upon starting robotInterface though, there are errors, related to the HALL sensor.
Interestingly, the same issue happens to the right arm also.

I have checked most connections I could think of, and they seem fine (also checking for continuity).

I attach the logs here, one with the left arm active and the right arm disabled and the other one with the right arm active and the left arm disabled.

[OsakaLogRightArm.txt](https://github.com/robotology/icub-support/files/1247936/OsakaLogRightArm.txt)
[OsakaLogLeftArm.txt](https://github.com/robotology/icub-support/files/1247937/OsakaLogLeftArm.txt)

Any suggestion on how to proceed?

julijenv, 2017-08-24:
Hi @kt10aan ,
For the OsakaLofRightArm- I guess that you remove all the part of the Left from the config file icub_all.xml, right?
ANyhow, On this log it is well deterministic that your only error is

> [ERROR]sharedcan [2] board 1 HALL SENSOR ERROR AXIS 0

So the only way to get rid of this message is to check the HALL SENSOR of the joint 0 of board 2B1 - if I'm not mistaken - it is the connector on which the wires have colors: blue-green-yellow-orange-brown.
Try to remove it from the header and relaunch the yarprobotInterface and see if the message error is still the same or if something else popup. (board 0 instead of 1 - or axis 1 instead of 0).
IF you pin out the good connector just check the wiring and if you cant see any broken wire just recrimp evrything.
Let me know.

[image]

For OsakaLeftARm I just can see so much of 

> [ERROR]recv buffer overrun ( 500 >  500)
[ERROR]run()-pushReadMsg() failed on CAN bus 1

and afterwards just tons of 

> [ERROR]sharedcan [1] board 1 HALL SENSOR ERROR AXIS 0
[ERROR]sharedcan [1] board 1 HALL SENSOR ERROR AXIS 1

It sounds quite strange to my hear that both HALL SENSORS on board 1B1 are broken but check it all the same, but what could really be a problem is the can Line that you just put back and that could bring bad communication on the line could you check the grounding correctly,
thx in advance

always available on skype for thoroughful debugging : skypie-ID is icub.support

kt10aan, 2017-08-24:
@julijenv,

Thanks! I will follow your instructions in the morning. However, could we have a Skype meeting in the morning from 10am onwards  (Italian time) to check things together, as there is an urgent need to have at least one arm operational start of next week for scheduled experiments?

julijenv, 2017-08-24:
@kt10aan , sure I will be there, C U tomorrow morning

kt10aan, 2017-08-25:
Second log for right arm.

[Osaka01RightArmLog2.txt](https://github.com/robotology/icub-support/files/1251660/Osaka01RightArmLog2.txt)

kt10aan, 2017-09-21:
@julijenv , as discussed yesterday, here are the results of the motor tests. All seem good :)

We checked all 8 motors on the BLLs on the back.

Checked for resistance between pairs of phases (black/white/red) and also between black/white/red and the body of the robot.

All 8 motors, between pairs of phases, have a resistance between 2.6Ω and 6.3Ω (with most of them being between 3~4.5Ω.

In all 8 motors, each phase to the body, no beeping. 

So I assume motors are healthy.
Email will follow with the rest of what we discussed :)
Many thanks!

julijenv, 2017-10-05:
Hi Kostas,
As you will have seen on mail and maybe if you have a look on skype, I need pictures with location of your worries (not only on the matter of the shoulder but, in the back connector and cables that you were mentionning as short) to estimate the good amount of replacement parts that I would need to bring to do a proper repair on site.
Please do them quickly, it is vital, thx in advance

closed 2018-01-16 by julijenv